Xilinx/Vitis-AI #524: Wrong include directories in library CMakeLists.txt

andreasmuetzel-ametek, 2021-09-03:
I tried to build a C++ application using the Vitis AI library (1.4) with a self-built petalinux sdk. When using CMake to find and link to the Vitis AI, CMake complains that the include path `/test_publ` does not exist. It looks like this stems from this line in the Vitis-AI-Library:

https://github.com/Xilinx/Vitis-AI/blob/f61061eef7550d98bf02a171604c9a9f283a7c47/tools/Vitis-AI-Library/model_config/CMakeLists.txt#L38-L40

This seems to be a bug in the usage of the generator expression. The expression `$<BUILD_INTERFACE:${CMAKE_CURRENT_BINARY_DIR}>/test_publ` resolves to `/test_publ` if `BUILD_INTERFACE` is false, which is the case when using the built library from a SDK. Shouldn't it be `$<BUILD_INTERFACE:${CMAKE_CURRENT_BINARY_DIR}/test_publ>`, which would result in an empty string in that case?

yongshengxu-xlnx, 2021-09-18:
Thank you for your feedback, please change PUBLIC to PRIVATE, we will fix this problem in the next update

yongshengxu-xlnx, 2022-02-16:
Hi, @andreasmuetzel-ametek . It has been fixed in 2.0, please have a test, after the test is ok, please close the issue. Thanks for your feedback.
And if there is no feedback, we will close the issue 1 week later.

andreasmuetzel-ametek, 2022-02-16:
Hi, thanks! I can't test it right now, but I had a look at the changed lines in the master branch and they look fine to me now. So I think it's reasonable to close the issue :)

closed 2022-02-16 by andreasmuetzel-ametek