Xilinx/Vitis-AI #525: DPUCADF8H examples for u200 does not work

HyukjunKwon, 2021-09-06:
Using vitis ai docker 1.4 

I'm following steps on https://github.com/Xilinx/Vitis-AI/tree/master/examples/DPUCADF8H/tf_resnet

Compiling seems fine, but the example does not work. 

compile result message:

 #############################################
 ######  Garbage 0
 #############################################
[UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
[UNILOG][INFO] Compile done.
[UNILOG][INFO] The meta json is saved to "/workspace/examples/DPUCADF8H/tf_resnet/out/meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/examples/DPUCADF8H/tf_resnet/out/resnet50_v1_tf_compiled.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is f7bba6e8f3243b803a2111276daf0075, and has been saved to "/workspace/examples/DPUCADF8H/tf_resnet/out/md5sum.txt"

but the example code does not work

error message:

WARNING: Logging before InitGoogleLogging() is written to STDERR
I0906 06:53:19.451807   922 main.cc:293] create running for subgraph: subgraph_resnet_v1_50/block1/unit_1/bottleneck_v1/add
terminate called after throwing an instance of 'std::runtime_error'
  what():  Error: Could not acquire CU
Aborted (core dumped)

*checked and restarted xbutler 
*other examples like inception_v1 has same error

simplelins, 2021-09-07:
sudo service xbutler restart

HyukjunKwon, 2021-09-09:
> sudo service xbutler restart
already tried restarting xbutler as I said, it does not work

SuZhan0322, 2022-09-01:
Hi @HyukjunKwon , Hope this issue has been resolved. If cold reboot and xbutler reset doesn't solve the issue.  It may be caused by the mismatch between xclbin and xmodel version.

SuZhan0322, 2022-09-01:
Please reopen this issue if you still have any other question. Thanks!

closed 2022-09-01 by SuZhan0322