Xilinx/Vitis-AI #527: vai_c_tensorflow2 compilation error for erfnet on Alveo U250

hleblevec, 2021-09-07:
Hi,

I'm having troubles compiling the tf2_erfnet_cityscapes_512_1024_54G_1.4 model from the AI Model Zoo for DPUCADF8H for Alveo U250. I downloaded the quantized model from the Github.

The command I'm using is :
`vai_c_tensorflow2 -m quantized/quantized.h5 -a /opt/vitis_ai/compiler/arch/DPUCADF8H/U250/arch.json -o compiled_output/ -n erfnet --options '{"input_shape": "4,512,1024,3"}'`

During the compilation, I am experiencing the following error:

quant_conv2d_transpose
'xir.Subgraph' object has no attribute 'create_op'
Edge by fake identity using the fix neuron layer quant_conv2d_transpose-conv-channel
adding arch quant_conv2d_transpose-conv-channel-quant_add_13 quant_conv2d_transpose-conv-channel quant_add_13 0 Name quant_add_13 Type conv2d-eltwise Composed ['quant_conv2d_58'] 
	Pred ['quant_conv2d_57', 'quant_conv2d_transpose-conv-channel'] 
	Succ ['quant_add_14', 'quant_conv2d_59'] 
adding arch quant_conv2d_transpose-conv-channel-quant_conv2d_55 quant_conv2d_transpose-conv-channel quant_conv2d_55 0 Name quant_conv2d_55 Type conv2d Composed [] 
	Pred ['quant_conv2d_transpose-conv-channel'] 
	Succ ['quant_conv2d_56'] 
terminate called after throwing an instance of 'pybind11::error_already_set'
  what():  KeyError: ('quant_conv2d_transpose',)

At:
  /opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/SC/Scheduler/GraphModifications/passes.py(1982): transpose_deconvolution_into_upsample_and_convolution
  /opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/SC/Scheduler/parallelism_original.py(2146): basic_graph
  /opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/SC/Scheduler/parallelism_original.py(2223): inception_and_lean_DAG
  /opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/SC/HwAbstraction/code_generation.py(586): explore_code_generation
  /opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/SC/HwAbstraction/code_generation.py(849): compile_sc_dpuv3int8
  /opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/SC/compiler_func.py(86): compile_sc_dpuv3int8_default
  /opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/SC/compiler_func.py(91): compile_subgraph_inplace

I have been able to compile other models for U250 before, and I don't understand what's wrong with this one. I'm also able to compile it for other targets like the U50. I hope you can help!

Thanks in advance!

paolodalberto, 2022-07-28:
The compiler indeed cannot work this example at all. 
let me see if I go back to it and check

paolodalberto, 2022-07-29:
I can compile it finally.
 will be working a binary comparison ...

paolodalberto, 2022-08-13:
good news and bad news. 
I could make it work and I was almost there but for dilated convolutions: 
there are 8x8 and 16x16 dilations: we transform the dilated into larger kernels unfortunately the 17x1, 1x17, 33x1, and 1x33 cannot be executed in HW .... 
ARG this was such a nice network !