Xilinx/Vitis-AI #528: vai_c_xir compile error with pytorch for densenet

liusiping10, 2021-09-08:
vai_c_xir -x quantize_result/DenseNet_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json -o outputs -n densenet121
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Graph name: DenseNet, with op num: 1175
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][XCOM_ITEM_UNDEFINED][The requested item is not finded or defined!] SmfSpec DenseNet__DenseNet_Sequential_features___DenseBlock_denseblock_2___DenseLayer_denselayer_10__input_97_fix_reshaped_inserted_fix_622 is not defined.
*** Check failure stack trace: ***
Aborted (core dumped)

qianglin-xlnx, 2021-09-10:
@liusiping10 
Could you please share the quantized model so we can analyze the issue? Thank you.

liusiping10, 2021-09-11:
[quantize_result-densenet121.zip](https://github.com/Xilinx/Vitis-AI/files/7147130/quantize_result-densenet121.zip)

Thank you for quick repsonse. Such error is caused by densenet121, but densenet161 also has similar compilation error.

I find that when the denselayer number >= 9 in one denseblock, a similar compilation error will occur.

for example:

when I define model with 

(1) _densenet('densenet121', 32, (6, 8, 24, 16), 64, pretrained, progress, **kwargs)

[UNILOG][FATAL][XCOM_ITEM_UNDEFINED][The requested item is not finded or defined!] SmfSpec DenseNet__DenseNet_Sequential_features___DenseBlock_denseblock_3___DenseLayer_denselayer_10__input_148_fix_reshaped_inserted_fix_594 is not defined.

(2) _densenet('densenet121', 32, (6, 8, 8, 16), 64, pretrained, progress, **kwargs)

[UNILOG][FATAL][XCOM_ITEM_UNDEFINED][The requested item is not finded or defined!] SmfSpec DenseNet__DenseNet_Sequential_features___DenseBlock_denseblock_4___DenseLayer_denselayer_10__input_199_fix_reshaped_inserted_fix_332 is not defined

(3) _densenet('densenet121', 32, (6, 8, 8, 8), 64, pretrained, progress, **kwargs)

[UNILOG][INFO] Compile done.

compiled successfully.

closed 2022-02-25 by qianglin-xlnx