Xilinx/Vitis-AI #530: import caffe failure in docker

FurballClaire, 2021-09-09:
Hi,

I'm trying to run the training process with a caffe model (plate recognition) by following the readme steps.  As in the readme, before generating lmdb, I need to use createLableLMDB.py to generate label file, but I got this error while import caffe.

python createLableLMDB.py --train-gt-file "../../data/train_gt.txt" --val-gt-file "../../data/val_gt.txt"
Traceback (most recent call last):
File "createLableLMDB.py", line 53, in <module>
import caffe
File "../../../caffe-xilinx/python/caffe/__init__.py", line 1, in <module>
from .pycaffe import Net, SGDSolver, NesterovSolver, AdaGradSolver, RMSPropSolver, AdaDeltaSolver, AdamSolver
File "../../../caffe-xilinx/python/caffe/pycaffe.py", line 13, in <module>
from ._caffe import Net, SGDSolver, NesterovSolver, AdaGradSolver, \
ModuleNotFoundError: No module named 'caffe._caffe'

 

I'm using docker so I thought I don't need to build caffe again, and even I did try to build it, I also got error

(vitis-ai-caffe) Vitis-AI /workspace/tutorials/caffe-xilinx > make pycaffe
CXX .build_release/src/caffe/proto/quantize.pb.cc
In file included from .build_release/src/caffe/proto/quantize.pb.cc:4:
.build_release/src/caffe/proto/quantize.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
10 | #include <google/protobuf/port_def.inc>
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:616: recipe for target '.build_release/src/caffe/proto/quantize.pb.o' failed
make: *** [.build_release/src/caffe/proto/quantize.pb.o] Error 1

 

Anyone know what cause this error ? 

Is it related to PYTHONPATH and how should I set this? (I google and found some people have same problem and it was related to PYTHONPATH export)

lji72, 2021-09-17:
Can you share with me the used docker version or link?

wangxd-xlnx, 2022-02-22:
Hi @FurballClaire 

Has this issue been resolved? Or is there anything else need help?

FurballClaire, 2022-02-22:
Hi ,

It's been solved , thanks

closed 2022-02-22 by qianglin-xlnx