Xilinx/Vitis-AI #531: caffe-xilinx/build/tools/caffe.bin does not exist

FurballClaire, 2021-09-09:
Hi ,
I try to train plate detection model with my own dataset (cf_plate-detection_320_320_0.49G_1.4), but seems caffe-xilinx has some missing files, it can't find caffe.bin

$ cd code/train
$ bash train.sh

(vitis-ai-caffe) Vitis-AI /workspace/tutorials/cf_plate-detection_320_320_0.49G_1.4 > cd code/train/
(vitis-ai-caffe) Vitis-AI /workspace/tutorials/cf_plate-detection_320_320_0.49G_1.4/code/train > bash train.sh
../../../caffe-xilinx/build/tools/caffe.bin does not exist, try use path in pre-build docker
F0909 09:40:41.270013   105 gpu_memory.cpp:171] Check failed: error == cudaSuccess (10 vs. 0)  invalid device ordinal
*** Check failure stack trace: ***
    @     0x7fb489f4a4dd  google::LogMessage::Fail()
    @     0x7fb489f52071  google::LogMessage::SendToLog()
    @     0x7fb489f49ecd  google::LogMessage::Flush()
    @     0x7fb489f4b76a  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fb488bfb135  caffe::GPUMemory::Manager::update_dev_info()
    @     0x7fb488bfb6af  caffe::GPUMemory::Manager::init()
    @     0x56372ad690ed  train()
    @     0x56372ad62a59  main

The caffe-xilinx/ folder was copied from models/AI-Model-Zoo/caffe-xilinx/
Do I need to do anything prior? As said in [this](https://github.com/Xilinx/Vitis-AI-Tutorials/tree/master/Design_Tutorials/01-caffe_cats_vs_dogs) cat&dog tutorial, I just need to copy it and place it in the same folder as my project and don't need to compile or build it, is that right?

kewang-xlnx, 2021-09-17:
Hi,
In the pre-build docker, there is not executable program “caffe.bin". It was replaced by "caffe". The "caffe.bin" should changed to "caffe" in your own shell script.
And the CUDA error may be because there are not enough GPU cores on your own machine.

niuxjxlnx, 2022-02-21:
Suppose the question has been answered and will close this issue if no more response.

closed 2022-02-28 by niuxjxlnx