Xilinx/Vitis-AI #532: vai_q_pytorch generates invalid Python code

kanpachi1, 2021-09-13:
`vai_q_pytorch` generates invalid variable names when a float model contains `torch.nn.Sequential`.

Reproducible example:
```
from pytorch_nndct.apis import torch_quantizer
import torch

class Model(torch.nn.Module):
    def __init__(self):
        super().__init__()
        self.weight = torch.nn.Parameter(torch.rand(1))
    def forward(self, input):
        return self.weight * input

model = torch.nn.Sequential(Model())
input = torch.rand(1)
quantizer = torch_quantizer('calib', model, input)
```
Output (Tested on the latest `xilinx/vitis-ai-cpu` image):
```
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'

[VAIQ_NOTE]: Loading NNDCT kernels...

[VAIQ_WARN]: CUDA is not available, change device to CPU

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cpu'.

[VAIQ_NOTE]: =>Parsing Sequential...

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/Sequential.py)
Traceback (most recent call last):
  File "test_sequential.py", line 13, in <module>
    quantizer = torch_quantizer('calib', model, input)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 77, in __init__
    custom_quant_ops = custom_quant_ops)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 122, in __init__
    device=device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 178, in prepare_quantizable_module
    quant_module = recreate_nndct_module(graph, True, export_file).to(device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 58, in recreate_nndct_module
    nndct_module = _reload_module(export_file, graph.name)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 50, in _reload_module
    spec.loader.exec_module(py_module)
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module
  File "<frozen importlib._bootstrap_external>", line 781, in get_code
  File "<frozen importlib._bootstrap_external>", line 741, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "quantize_result/Sequential.py", line 10
    self.0_weight = torch.nn.parameter.Parameter(torch.Tensor(1,))
          ^
SyntaxError: invalid token
```

`self.0_weight` is an invalid variable name because Python variables can't start with digits.

There is a workaround for this issue. Initializing `torch.nn.Sequential` with `OrderedDict` can replace `0` with an arbitrary name:
```
from collections import OrderedDict
model = torch.nn.Sequential(OrderedDict({'a': Model()}))
quantizer = torch_quantizer('calib', model, input) # no SyntaxError
```

carrotandjzy, 2021-09-17:
So far, the free parameters(self.weight=torch.nn.Parameter(...) ) is not supported in vai_q_pytorch offically. You can use some operations(conv, linear ...) which have bounded parameters instead.

kanpachi1, 2021-09-17:
> So far, the free parameters(self.weight=torch.nn.Parameter(...) ) is not supported in vai_q_pytorch officially.

Thank you for your response. That's a shame😞

I found another similar problem related to `torch.nn.Parameter`, and so I leave it here for someone facing the same problem before closing the issue.
```
from pytorch_nndct.apis import torch_quantizer
import torch

class Model(torch.nn.Module):
    def __init__(self):
        super().__init__()
        self.param = torch.nn.Parameter(torch.tensor(1.))
    def forward(self, input):
        return input, self.param

model = Model()
input = torch.ones(1)
quantizer = torch_quantizer('calib', model, input)
```

```
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'

[VAIQ_NOTE]: Loading NNDCT kernels...

[VAIQ_WARN]: CUDA is not available, change device to CPU

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cpu'.

[VAIQ_NOTE]: =>Parsing Model...

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/Model.py)
Traceback (most recent call last):
  File "test_parameter.py", line 13, in <module>
    quantizer = torch_quantizer('calib', model, input)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 77, in __init__
    custom_quant_ops = custom_quant_ops)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 122, in __init__
    device=device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 178, in prepare_quantizable_module
    quant_module = recreate_nndct_module(graph, True, export_file).to(device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 58, in recreate_nndct_module
    nndct_module = _reload_module(export_file, graph.name)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 50, in _reload_module
    spec.loader.exec_module(py_module)
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module
  File "<frozen importlib._bootstrap_external>", line 781, in get_code
  File "<frozen importlib._bootstrap_external>", line 741, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "quantize_result/Model.py", line 12
    return self.output_module_0,Model::param
                                     ^
SyntaxError: invalid syntax
```
An invalid variable name is generated when return values of `forward` contain a `torch.nn.Parameter`.

closed 2021-09-17 by kanpachi1