Xilinx/Vitis-AI #533: vai_q_pytorch causes IndexError

kanpachi1, 2021-09-13:
vai_q_pytorch causes IndexError when I use `torch.Tensor.size`,`torch.Tensor.permute` and `torch.Tensor.view`.

Reproducible example:
```
from pytorch_nndct.apis import torch_quantizer
import torch

class ModelA(torch.nn.Module):
    def __init__(self):
        super().__init__()
        self.fc = torch.nn.Linear(2, 2)
    def forward(self, input):
        batch = input.size(0)
        output = self.fc(input.permute(0, 2, 1))
        return output.view(batch, -1)

model = ModelA()
input = torch.rand(1, 2, 3)
quantizer = torch_quantizer('calib', model, input)
```

Output (Tested on the latest `xilinx/vitis-ai-cpu` image):
```
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'

[VAIQ_NOTE]: Loading NNDCT kernels...

[VAIQ_WARN]: CUDA is not available, change device to CPU

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cpu'.

[VAIQ_NOTE]: =>Parsing ModelA...

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/ModelA.py)
Traceback (most recent call last):
  File "test_view.py", line 24, in <module>
    quantizer = torch_quantizer('calib', model, input)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 77, in __init__
    custom_quant_ops = custom_quant_ops)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 130, in __init__
    quantizer.setup(graph, False, lstm_app, custom_quant_ops=custom_quant_ops)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/quantization/base_quantizer.py", line 82, in setup
    custom_quant_ops=custom_quant_ops)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/quantization/quant_info.py", line 42, in __init__
    self._quant_info = quant_strategy.create_quant_config(self)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/quantization/quant_strategy.py", line 138, in create_quant_config
    return self._get_default_quant_config(quant_info_mgr)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/quantization/quant_strategy.py", line 113, in _get_default_quant_config
    end = quant_info_mgr.quant_output(p_n.name).name
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/nndctgraph_holder.py", line 250, in quant_output
    end_node = self.__Nndctgraph.node(self._QuantGroups[node.name][idx])
IndexError: list index out of range
```

I'm not sure why but no error raised without any one of `torch.Tensor.size`,`torch.Tensor.permute` and `torch.Tensor.view`.
```
class ModelB(torch.nn.Module):
    def __init__(self):
        super().__init__()
        self.fc = torch.nn.Linear(2, 2)
    def forward(self, input):
        batch = 1
        output = self.fc(input.permute(0, 2, 1))
        return output.view(batch, -1)
class ModelC(torch.nn.Module):
    def __init__(self):
        super().__init__()
        self.fc = torch.nn.Linear(2, 2)
    def forward(self, input):
        batch = input.size(0)
        output = self.fc(input)
        return output.view(batch, -1)
class ModelD(torch.nn.Module):
    def __init__(self):
        super().__init__()
        self.fc = torch.nn.Linear(2, 2)
    def forward(self, input):
        batch = input.size(0)
        output = self.fc(input.permute(0, 2, 1))
        return output
model = ModelB()
input = torch.rand(1, 2, 3)
quantizer = torch_quantizer('calib', model, input) # No IndexError
model = ModelC()
input = torch.rand(1, 3, 2)
quantizer = torch_quantizer('calib', model, input) # No IndexError
model = ModelD()
input = torch.rand(1, 2, 3)
quantizer = torch_quantizer('calib', model, input) # No IndexError
```

niuxjxlnx, 2021-09-17:
The error is caused by a bug, which willed be fixed in coming vitis-ai 1.4.1.

kanpachi1, 2021-09-17:
Thank you for looking into this. I'm looking forward to the next release!

closed 2021-09-17 by kanpachi1