Xilinx/Vitis-AI #534: meet issue "pip3: command not found" of using  "xilinx-zcu102-dpu-v2021.1-v1.4.0.img.gz"

chayitw, 2021-09-14:
Hi,
  When following the step of the "Running on Zynq" tutorial, i download the Petalinux image for 
  our target: ZCU-102 and could boot successful. But when i try to install dependency as descripted in "petalinux setup script"(petalinux_setup.sh),  i meet the issue: "pip3: command not found".

  Maybe something wrong for the link to the image: "xilinx-zcu102-dpu-v2021.1-v1.4.0.img.gz", just guess, not sure ...

  I also have a ZCU-104 board, and try "xilinx-zcu104-dpu-v2021.1-v1.4.0.img.gz", it work well.

  Any suggestion is appreciated!
Thanks,
-chayi

The boot info is as below:

Xilinx Zynq MP First Stage Boot Loader
Release 2021.1   Jun  6 2021  -  07:07:32
NOTICE:  ATF running on XCZU9EG/silicon v4/RTL5.1 at 0xfffea000
NOTICE:  BL31: v2.4(release):v1.1-7609-g851523ea2
NOTICE:  BL31: Built : 08:27:07, Apr 28 2021

U-Boot 2021.01 (Jun 01 2021 - 11:54:06 +0000)

Model: ZynqMP ZCU102 Rev1.0
Board: Xilinx ZynqMP
DRAM:  4 GiB
PMUFW:  v1.1
EL Level:       EL2
Chip ID:        zu9eg
NAND:  0 MiB
MMC:   mmc@ff170000: 0
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Bootmode: LVL_SHFT_SD_MODE1
Reset reason:   EXTERNAL
Net:
ZYNQ GEM: ff0e0000, mdio bus ff0e0000, phyaddr 12, interface rgmii-id
eth0: ethernet@ff0e0000
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
2597 bytes read in 12 ms (210.9 KiB/s)
## Executing script at 20000000
Trying to load boot images from mmc0
27589120 bytes read in 2041 ms (12.9 MiB/s)
56499 bytes read in 16 ms (3.4 MiB/s)
## Flattened Device Tree blob at 00100000
   Booting using the fdt blob at 0x100000
   Loading Device Tree to 000000007dcff000, end 000000007dd0fcb2 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.0-xilinx-v2021.1 (oe-user@oe-host) (aarch64-xi                                                                                 linx-linux-gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.1) #1 SMP Fri Jun 4 15:5                                                                                 7:16 UTC 2021
[    0.000000] Machine model: ZynqMP ZCU102 Rev1.0
[    0.000000] earlycon: cdns0 at MMIO 0x00000000ff000000 (options '115200n8')
[    0.000000] printk: bootconsole [cdns0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 1536 MiB at 0x000000001dc00000
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x000000087fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000007fefffff]
[    0.000000]   node   0: [mem 0x0000000800000000-0x000000087fffffff]
[... 386 lines cut ...]
root@xilinx-zcu102-2021_1:~# D-BUS per-session daemon address is: unix:abstract=/tmp/dbus-UYgDwZQgYC,guid=4b8a1061a101fe6bcd1b9fcd6015d998
matchbox: Cant find a keycode for keysym 269025056
matchbox: ignoring key shortcut XF86Calendar=!$contacts

matchbox: Cant find a keycode for keysym 2809
matchbox: ignoring key shortcut telephone=!$dates

matchbox: Cant find a keycode for keysym 269025050
matchbox: ignoring key shortcut XF86Start=!matchbox-remote -desktop

[settings daemon] Forking. run with -n to prevent fork
dbus-daemon[882]: Activating service name='org.a11y.atspi.Registry' requested by ':1.0' (uid=0 pid=878 comm="matchbox-panel --start-applets showdesktop,windows")
dbus-daemon[882]: Successfully activated service 'org.a11y.atspi.Registry'
SpiRegistry daemon is running with well-known name - org.a11y.atspi.Registry

** (matchbox-desktop:877): WARNING **: 22:11:38.516: Error loading icon: Icon 'terminal' not present in theme Sato

root@xilinx-zcu102-2021_1:~#
root@xilinx-zcu102-2021_1:~#
root@xilinx-zcu102-2021_1:~#
root@xilinx-zcu102-2021_1:~# pip3
-sh: pip3: command not found
root@xilinx-zcu102-2021_1:~#
root@xilinx-zcu102-2021_1:~# locate pip3
root@xilinx-zcu102-2021_1:~#

runner42195, 2021-09-17:
I meet the same issue, did you figure out it.

qianglin-xlnx, 2021-09-17:
Hi @chayitw 
Thank you for your feedback. We will check this issue and feedback to you.

tianfang-fafafa, 2021-09-18:
hi @chayitw  first, I don't who release "petalinux setup script"(petalinux_setup.sh).   In a developer's view, the zcu104 supports package gstreamer-vcu-examples which depends on pip3, so the pip3 would be installed autometically. So the zcu102 would have no pip3.  can you please try "dnf install python3-pip" ?

tianfang-fafafa, 2021-09-18:
hi @runner42195 , please look into previous reply

chayitw, 2021-09-18:
hi @tianfang-fafafa , i have holiday until 9/14, and no board in hand. 
Thanks for your suggestion, i will try it when back to my office.

hi @qianglin-xlnx, thank for your kindly reply also ^ ^ 

( Happy Moon Festival ^ ^ )

chayitw, 2021-09-22:
hi @tianfang-fafafa, 
    I just try "dnf install python3-pip" on ZCU102, and meet the below problem:

    1. Firstly, i meet the issue about "404" for the download path such like that:
        Errors during downloading metadata for repository 'oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmpeg':
  - Status code: 404 for http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmpeg/repodata/repomd.xml (IP: 23.41.131.204)
Error: Failed to download metadata for repo 'oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmpeg': Cannot download repomd.xml: Cannot download repodata/repomd.xml: All mirrors were tried
        ...

        I find that the download path in original /etc/yum.repos.d/oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm.repo maybe old, so i change some path as below:

       root@xilinx-zcu102-2021_1:/etc/yum.repos.d# diff oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm.repo      oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm.repo.org
18c18
< baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmp_eg
---
> baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmpeg
23c23
< baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmp_ev
---
> baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmpev
28c28
< baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmp_dr
---
> baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmpdr
33c33
< baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmp_cg
---
> baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmpcg

2. Secondly, it could download after above adjust but meet the issue about "No match for argument: python3-pip":
   
root@xilinx-zcu102-2021_1:/etc/yum.repos.d# dnf install python3-pip
OE Remote Repo: sswreleases rel-v2021.1 generic rpm noarch                                                       6.2 kB/s | 3.0 kB     00:00
OE Remote Repo: sswreleases rel-v2021.1 generic rpm aarch64                                                      7.1 kB/s | 3.0 kB     00:00
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmp                                                       6.7 kB/s | 3.0 kB     00:00
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpeg                                                     187 kB/s | 294 kB     00:01
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpev                                                     400 kB/s | 296 kB     00:00
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpdr                                                     6.0 kB/s |  11 kB     00:01
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpcg                                                     3.7 kB/s | 5.6 kB     00:01
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmp_generic                                               6.2 kB/s | 3.0 kB     00:00
No match for argument: python3-pip
Error: Unable to find a match: python3-pip

Need your suggestion to solve this issue, thanks. 

br,
--chayi

tianfang-fafafa, 2021-09-23:
@chayitw  ok. I will reply you once solving this issue

tianfang-fafafa, 2021-09-23:
@chayitw  can you please try baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/cortexa72_cortexa53/ ?
my board can't access to Internent in my environment

chayitw, 2021-09-24:
hi @tianfang-fafafa, thanks! It seems work after change the baseurl to be "http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/cortexa72_cortexa53/".

The detail info. is as below: ( just need 2 steps suggested by @tianfang-fafafa )

1. change the baseurl of "/etc/yum.repos.d/oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm.repo":

root@xilinx-zcu102-2021_1:/etc/yum.repos.d# cat /etc/yum.repos.d/oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm.repo
[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-noarch]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm noarch
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/cortexa72_cortexa53
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-aarch64]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm aarch64
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/cortexa72_cortexa53
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmp]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmp
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/cortexa72_cortexa53
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmpeg]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpeg
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/cortexa72_cortexa53
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmpev]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpev
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/cortexa72_cortexa53
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmpdr]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpdr
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/cortexa72_cortexa53
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmpcg]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpcg
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/cortexa72_cortexa53
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmp_generic]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmp_generic
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/cortexa72_cortexa53
gpgcheck=0

2. using "dnf install python3-pip" to install pip3:

root@xilinx-zcu102-2021_1:/etc/yum.repos.d# dnf install python3-pip
OE Remote Repo: sswreleases rel-v2021.1 generic rpm noarch                467 kB/s | 3.8 MB     00:08
OE Remote Repo: sswreleases rel-v2021.1 generic rpm aarch64                18 MB/s | 3.8 MB     00:00
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmp                 16 MB/s | 3.8 MB     00:00
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpeg               18 MB/s | 3.8 MB     00:00
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpev               18 MB/s | 3.8 MB     00:00
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpdr              1.4 MB/s | 3.8 MB     00:02
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpcg               15 MB/s | 3.8 MB     00:00
OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmp_generic         18 MB/s | 3.8 MB     00:00
Last metadata expiration check: 0:00:05 ago on Wed Feb 10 03:58:09 2021.
Dependencies resolved.
==========================================================================================================
 Package  Architecture        Version     Repository                                                 Size
==========================================================================================================
Installing:
 python3-pip
          cortexa72_cortexa53 20.0.2-r0.0 oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-noarch 1.6 M
Installing dependencies:
 python3-pkg-resources
          cortexa72_cortexa53 49.6.0-r0.0 oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-noarch 203 k
 python3-setuptools
[... 22 lines cut ...]
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                  1/1
  Installing       : python3-setuptools-lic-49.6.0-r0.0.cortexa72_cortexa53                           1/5
  Installing       : python3-pkg-resources-49.6.0-r0.0.cortexa72_cortexa53                            2/5
  Installing       : python3-setuptools-49.6.0-r0.0.cortexa72_cortexa53                               3/5
  Installing       : python3-pip-lic-20.0.2-r0.0.cortexa72_cortexa53                                  4/5
  Installing       : python3-pip-20.0.2-r0.0.cortexa72_cortexa53                                      5/5
  Verifying        : python3-pip-20.0.2-r0.0.cortexa72_cortexa53                                      1/5
  Verifying        : python3-pip-lic-20.0.2-r0.0.cortexa72_cortexa53                                  2/5
  Verifying        : python3-pkg-resources-49.6.0-r0.0.cortexa72_cortexa53                            3/5
  Verifying        : python3-setuptools-49.6.0-r0.0.cortexa72_cortexa53                               4/5
  Verifying        : python3-setuptools-lic-49.6.0-r0.0.cortexa72_cortexa53                           5/5

Installed:
  python3-pip-20.0.2-r0.0.cortexa72_cortexa53
  python3-pip-lic-20.0.2-r0.0.cortexa72_cortexa53
  python3-pkg-resources-49.6.0-r0.0.cortexa72_cortexa53
  python3-setuptools-49.6.0-r0.0.cortexa72_cortexa53
  python3-setuptools-lic-49.6.0-r0.0.cortexa72_cortexa53

Complete!

tianfang-fafafa, 2021-09-25:
@chayitw  hi，can you please provide me the default /etc/yum.repos.d/oe-xxx.repo ?

chayitw, 2021-09-27:
@tianfang-fafafa, thanks, Yes, of course, 
the default "/etc/yum.repos.d/oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm.repo" is as below:

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-noarch]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm noarch
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/noarch
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-aarch64]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm aarch64
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/aarch64
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmp]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmp
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmp
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmpeg]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpeg
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmpeg
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmpev]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpev
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmpev
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmpdr]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpdr
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmpdr
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmpcg]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmpcg
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmpcg
gpgcheck=0

[oe-remote-repo-sswreleases-rel-v2021.1-generic-rpm-zynqmp_generic]
name=OE Remote Repo: sswreleases rel-v2021.1 generic rpm zynqmp_generic
baseurl=http://petalinux.xilinx.com/sswreleases/rel-v2021.1/generic/rpm/zynqmp_generic
gpgcheck=0

tianfang-fafafa, 2021-09-27:
@chayitw  thanks