Xilinx/Vitis-AI #536: pytorch model compilation error: Unregistered attribute

waldenou, 2021-09-17:
Hi,

While I was trying to compile for a pytorch model using xir. I have encountered the following error:
```
(vitis-ai-pytorch) Vitis-AI /workspace/test/braggNN > vai_c_xir -x quantize_result/BraggNN_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCADF8H/U250/arch.json -o compiled -n braggnn
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][INFO] Target architecture: DPUCADF8H_ISA0
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCADF8H_ISA0
[UNILOG][INFO] Graph name: BraggNN, with op num: 121
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][XIR_UNREGISTERED_ATTR][Unregistered attribute!] Attrs doesn't contain attribute data
*** Check failure stack trace: ***
Aborted (core dumped)
```

I'm not sure what causes this error. It seems that the compiler was unable to find the 'data' key in the required attributes of Op type 'const'.

I have attached quantize results, xmodel.txt and related model files. Any help is appreciated! Thanks!
[test_files.zip](https://github.com/Xilinx/Vitis-AI/files/7186426/test_files.zip)

shua1zhang, 2021-09-21:
Hi @waldenou ,

Which Vitis AI version are you using and have you tested fully on VAI 1.4? 

We will check the files and feedback to you.

waldenou, 2021-09-22:
Hi @shua1zhang,

I'm using Vitis AI 1.4. Thank you.

waldenou, 2021-10-08:
Hi @shua1zhang,

Just wondering is there any update on this? Thanks!

SuZhan0322, 2022-09-01:
hi @waldenou ,  Is this issue solved?

qianglin-xlnx, 2023-01-17:
Closing since no activity for more than 4 months, please reopen if you still have any questions, thanks

closed 2023-01-17 by qianglin-xlnx