Xilinx/Vitis-AI #539: How to create mult-DPUs in vck190

fjzzlhc, 2021-09-24:
I have tried to create a DPUCVDX8G followed by the instructions of XVDPU-TRD project and succeeded in running it in the vck190 platform. However, it only have one DPU while I need more than one to run different algorithms in the same time (Algorithms can run in one dpu by scheduling , but I need to isolate them for avoiding bloking, just like ZCU102 which have 3 DPUs). The instructions don't tell me how to create multi-DPUs. So I wonder if I could do this?How to?

qianglin-xlnx, 2021-09-24:
Hi @fjzzlhc 
Currently, only one dpu core with different batch number is supported in vck190.

fjzzlhc, 2021-09-26:
> Hi @fjzzlhc
> Currently, only one dpu core with different batch number is supported in vck190.

Thank you for your reply. Besides, may I ask whether or not there is a plan to make it? And when?

qianglin-xlnx, 2021-09-26:
@fjzzlhc 
Yes. We are developing this feature internally. But I'm not sure when it will be public released.

shashwat1198, 2022-02-23:
> rator
> ### [[image]](/qianglin-xlnx) ** [qianglin-xlnx](/qianglin-xlnx) ** commented [on Sep 25, 2021](#issuecomment-927239199)

Hello,

I was wondering if this feature is something you guys are still working on? I want to do the same thing i.e run multiple DPUs at the same time on an Ultrascale+ FPGA  board( ZCU104 ). 

Any pointers or suggestions from you would be very helpful.
Shashwat

qianglin-xlnx, 2022-02-24:
Hi @shashwat1198 
For VCK190, we are working on this feature.
For ZCU104, multiple DPUs is supported. You can run different application on it. You can refer to https://github.com/Xilinx/Vitis-AI/tree/master/demo/Vitis-AI-Library/apps/seg_and_pose_detect

shashwat1198, 2022-02-25:
> Hi @shashwat1198 For VCK190, we are working on this feature. For ZCU104, multiple DPUs is supported. You can run different application on it. You can refer to https://github.com/Xilinx/Vitis-AI/tree/master/demo/Vitis-AI-Library/apps/seg_and_pose_detect

Hey,

Thank you for the response. Just one more question. Is this feature implemented only with C++ APIs? Is there any implementation that has used python APIs?

Kind regards,
Shashwat

qianglin-xlnx, 2022-02-27:
Hi @shashwat1198 
Both C++ and Python are supported. There is no limitation for this.
You can run 2 dpu each at a time. You can also run different models on each core at the same time.

shashwat1198, 2022-02-27:
> Hi @shashwat1198 Both C++ and Python are supported. There is no limitation for this. You can run 2 dpu each at a time. You can also run different models on each core at the same time.

Hey @qianglin-xlnx ,

Thank you so much for the clarification. It really helps a lot.

If it is possible can you point me to an implementation that uses python VART APIs for this purpose?

Kind regards,
Shashwat

qianglin-xlnx, 2022-03-03:
Hi @shashwat1198 
You can refer to : https://github.com/Xilinx/Vitis-AI/blob/master/demo/VART/resnet50_mt_py/resnet50.py

closed 2022-03-03 by qianglin-xlnx

shashwat1198, 2022-03-03:
> Hi @shashwat1198 You can refer to : https://github.com/Xilinx/Vitis-AI/blob/master/demo/VART/resnet50_mt_py/resnet50.py

Okay. Thank you for your reply. @qianglin-xlnx 

Hey I know that the DPU gets it's input data from the PS using VART APIs. 

Would it be possible that we can write data maybe on a FIFO or BRAM in the PL side and the DPU directly reads it from there instead of getting it from the PS? Is this something that can be achieved with the DPU?

Shashwat

shashwat1198, 2022-03-03:
> Hi @shashwat1198 You can refer to: #591 and #313

Thank you for your responses @qianglin-xlnx .

Kind regards,
Shashwat

ashan2000, 2023-09-12:
Hi, I'm using the Kria KV260 board. also trying to implement multiple DPUs to deploy multiple Xmodels simultaneously. yet a beginner in this field. Happy to receive some help in this regard if it is possible for you @qianglin-xlnx