Xilinx/Vitis-AI #544: use epsilon for floating-point equal comparison

hfkou, 2021-09-28:
In order to compare floating point numbers at the equal situation, how about consider using:
    if ( tmp < 0 && abs(tmp - floor(tmp) - 0.5) < (Dtype)FLT_EPSILON)
Other than:
    if ( tmp < 0 && ( tmp - floor( tmp ) ) == 0.5 )

Related files that above fix could apply:
~/Vitis-AI/models/AI-Model-Zoo/caffe-xilinx/src/caffe/util/quantize.cpp
~/Vitis-AI/models/AI-Model-Zoo/caffe-xilinx/src/caffe/util/quantize.cu

niuxjxlnx, 2022-02-22:
Hi, @hfkou:

Thanks for your suggestion. The code you suggested is more C.S style. But here 0.5(binary format is 0.1) can be absolutely expressed by binary number. So it will not cause any problem.

niuxjxlnx, 2022-02-24:
@hfkou:

The theory you showed is completely right for general float number comparison. But here, quantizer code is to simulate the INT8 behaviour on DPU, where always can get absolute comparison result.