Xilinx/Vitis-AI #548: An error of the process of converting to xmodel with Pytorch

qw85639229, 2021-10-03:
I'm trying to convert my model (Resnet-based) to xmodel and it's successfully quantized. But there happens an error when deploy.
'''
[VAIQ_NOTE]: Quantization test process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cpu'.

[VAIQ_NOTE]: =>Parsing ResNet...

[VAIQ_NOTE]: =>Doing weights equalization...
/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/optimization/commander.py:444: RuntimeWarning: divide by zero encountered in true_divide
  scale = np.where(sqrt_of_ranges != 0, range_0 / sqrt_of_ranges, scale)
/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/optimization/commander.py:444: RuntimeWarning: invalid value encountered in true_divide
  scale = np.where(sqrt_of_ranges != 0, range_0 / sqrt_of_ranges, scale)

[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/ResNet.py)

[VAIQ_NOTE]: =>Get module with quantization.

[VAIQ_NOTE]: =>Converting to xmodel ...
Traceback (most recent call last):
  File "test.py", line 202, in <module>
    main()
  File "test.py", line 59, in main
    main_worker(args)
  File "test.py", line 145, in main_worker
    quantizer.export_xmodel(deploy_check=False)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 98, in export_xmodel
    self.processor.export_xmodel(output_dir, deploy_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 165, in export_xmodel
    dump_xmodel(output_dir, deploy_check, self._lstm_app)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 188, in dump_xmodel
    deploy_graphs = get_deploy_graph_list(quantizer.quant_model, quantizer.Nndctgraph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 271, in get_deploy_graph_list
    g_optmizer.constant_folding()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/compile/deploy_optimizer.py", line 83, in constant_folding
    node.set_node_attr(attr_name, attr_val)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_node.py", line 209, in set_node_attr
    self._op.set_attr(key, value)  
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_operator.py", line 272, in set_attr
    self._attrs[attr_name].value = attr_value
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_operator.py", line 126, in value
    f"The type of attr '{self._name.value}' should be {self._type} instead of {type(value[0])}"
TypeError: The type of attr 'kernel' should be <class 'int'> instead of <class 'NoneType'>
'''

carrotandjzy, 2021-10-11:
@qw85639229  Can you provide model file and dummy inputs ?

dsp-avt, 2021-10-15:
I've seen this issue when I do not run the quantized model after running torch_quantizer in test mode.

Example that causes error:
```python
# Quantize the float model
resnet.eval()
cal_data = torch.randn(batch_size, 3, 160, 160)
cal_data = cal_data.to(device)
quantizer = torch_quantizer("test", resnet, (cal_data), output_dir="quant_out", device=device)
quant_model = quantizer.quant_model

# Generate the xmodel file
quantizer.export_xmodel(deploy_check=False, output_dir="quant_out")
```

ERROR:
```txt
[VAIQ_NOTE]: Quantization test process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cuda'.

[VAIQ_NOTE]: =>Parsing InceptionResnetV1...

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quant_out/InceptionResnetV1.py)

[VAIQ_NOTE]: =>Get module with quantization.

[VAIQ_NOTE]: =>Converting to xmodel ...

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-23-eccf641b156c> in <module>
      8 # Generate the xmodel file
      9 #temp = quant_model(cal_data)
---> 10 quantizer.export_xmodel(deploy_check=False, output_dir="quant_out")

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py in export_xmodel(self, output_dir, deploy_check)
     96   # export xmodel for compilation
     97   def export_xmodel(self, output_dir="quantize_result", deploy_check=False):
---> 98     self.processor.export_xmodel(output_dir, deploy_check)
     99 
    100   # deploy the trained model for quant aware training process

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py in export_xmodel(self, output_dir, deploy_check)
    163   # export xmodel file to be compiled for deployment
    164   def export_xmodel(self, output_dir="quantize_result", deploy_check=False):
--> 165     dump_xmodel(output_dir, deploy_check, self._lstm_app)
    166 
    167 

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py in dump_xmodel(output_dir, deploy_check, lstm_app)
    186 
    187     NndctScreenLogger().info("=>Converting to xmodel ...")
--> 188     deploy_graphs = get_deploy_graph_list(quantizer.quant_model, quantizer.Nndctgraph)
    189     depoly_infos = compiler.get_deloy_graph_infos(quantizer, deploy_graphs)
    190     if not lstm_app:

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py in get_deploy_graph_list(quant_model, nndct_graph)
    269   connect_module_with_graph(quant_model, nndct_graph, recover_param=False)
    270 
--> 271   g_optmizer.constant_folding()
    272   if NndctOption.nndct_parse_debug.value >= 3:
    273     NndctDebugLogger.write(f"\nfrozen dev graph:\n{g_optmizer.frozen_graph}")

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/compile/deploy_optimizer.py in constant_folding(self)
     81           if node.op.attrs[attr_name].type == list:
     82             attr_val = [attr_val]
---> 83           node.set_node_attr(attr_name, attr_val)
     84 
     85     if folding_nodes:

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_node.py in set_node_attr(self, key, value)
    207 
    208   def set_node_attr(self, key, value):
--> 209     self._op.set_attr(key, value)
    210 
    211   def node_config(self, key):

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_operator.py in set_attr(self, attr_name, attr_value)
    270   def set_attr(self, attr_name: str, attr_value: List[Any]) -> None:
    271     self._check_attr_valid(attr_name, attr_value)
--> 272     self._attrs[attr_name].value = attr_value
    273 
    274   def get_config(self, config_name: str) -> Any:

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_operator.py in value(self, value)
    124     if not isinstance(value[0], self._type):
    125       raise TypeError(
--> 126           f"The type of attr '{self._name.value}' should be {self._type} instead of {type(value[0])}"
    127       )
    128     if self._size is not None and len(value) != self._size:

TypeError: The type of attr 'kernel' should be <class 'int'> instead of <class 'NoneType'>
```

Running the model prior to export fixed the issue.  Example with update:
```python
# Quantize the float model
resnet.eval()
cal_data = torch.randn(batch_size, 3, 160, 160)
cal_data = cal_data.to(device)
quantizer = torch_quantizer("test", resnet, (cal_data), output_dir="quant_out", device=device)
quant_model = quantizer.quant_model

# Generate the xmodel file
temp = quant_model(cal_data)
quantizer.export_xmodel(deploy_check=False, output_dir="quant_out")
```

Notice the addition of line ``temp = quant_model(cal_data)`` prior to the export_xmodel command

Successful output with fixed code:
```txt

[VAIQ_NOTE]: Quantization test process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cuda'.

[VAIQ_NOTE]: =>Parsing InceptionResnetV1...

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quant_out/InceptionResnetV1.py)

[VAIQ_NOTE]: =>Get module with quantization.

[VAIQ_NOTE]: =>Converting to xmodel ...

[VAIQ_NOTE]: =>Successfully convert 'InceptionResnetV1' to xmodel.(quant_out/InceptionResnetV1_int.xmodel)

```

qw85639229, 2021-10-16:
> I've seen this issue when I do not run the quantized model after running torch_quantizer in test mode.
> 
> Example that causes error:
> 
> ```python
> # Quantize the float model
> resnet.eval()
> cal_data = torch.randn(batch_size, 3, 160, 160)
> cal_data = cal_data.to(device)
> quantizer = torch_quantizer("test", resnet, (cal_data), output_dir="quant_out", device=device)
> quant_model = quantizer.quant_model
> 
> # Generate the xmodel file
> quantizer.export_xmodel(deploy_check=False, output_dir="quant_out")
> ```
> 
> ERROR:
> 
> ```
> [VAIQ_NOTE]: Quantization test process start up...
> 
> [VAIQ_NOTE]: =>Quant Module is in 'cuda'.
> 
> [VAIQ_NOTE]: =>Parsing InceptionResnetV1...
> 
> [VAIQ_NOTE]: =>Doing weights equalization...
> 
> [VAIQ_NOTE]: =>Quantizable module is generated.(quant_out/InceptionResnetV1.py)
> 
> [VAIQ_NOTE]: =>Get module with quantization.
> 
> [VAIQ_NOTE]: =>Converting to xmodel ...
> 
> ---------------------------------------------------------------------------
> TypeError                                 Traceback (most recent call last)
> <ipython-input-23-eccf641b156c> in <module>
>       8 # Generate the xmodel file
>       9 #temp = quant_model(cal_data)
> ---> 10 quantizer.export_xmodel(deploy_check=False, output_dir="quant_out")
> 
> /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py in export_xmodel(self, output_dir, deploy_check)
>      96   # export xmodel for compilation
>      97   def export_xmodel(self, output_dir="quantize_result", deploy_check=False):
> ---> 98     self.processor.export_xmodel(output_dir, deploy_check)
>      99 
>     100   # deploy the trained model for quant aware training process
> 
> /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py in export_xmodel(self, output_dir, deploy_check)
>     163   # export xmodel file to be compiled for deployment
>     164   def export_xmodel(self, output_dir="quantize_result", deploy_check=False):
> --> 165     dump_xmodel(output_dir, deploy_check, self._lstm_app)
>     166 
>     167 
> 
> /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py in dump_xmodel(output_dir, deploy_check, lstm_app)
>     186 
>     187     NndctScreenLogger().info("=>Converting to xmodel ...")
> --> 188     deploy_graphs = get_deploy_graph_list(quantizer.quant_model, quantizer.Nndctgraph)
>     189     depoly_infos = compiler.get_deloy_graph_infos(quantizer, deploy_graphs)
>     190     if not lstm_app:
> 
> /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py in get_deploy_graph_list(quant_model, nndct_graph)
>     269   connect_module_with_graph(quant_model, nndct_graph, recover_param=False)
>     270 
> --> 271   g_optmizer.constant_folding()
>     272   if NndctOption.nndct_parse_debug.value >= 3:
>     273     NndctDebugLogger.write(f"\nfrozen dev graph:\n{g_optmizer.frozen_graph}")
> 
> /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/compile/deploy_optimizer.py in constant_folding(self)
>      81           if node.op.attrs[attr_name].type == list:
[... 98 lines cut ...]
Traceback (most recent call last):
  File "test.py", line 242, in <module>
    main()
  File "test.py", line 91, in main
    main_worker(args)
  File "test.py", line 160, in main_worker
    quantizer.export_xmodel(deploy_check=True)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 98, in export_xmodel
    self.processor.export_xmodel(output_dir, deploy_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 165, in export_xmodel
    dump_xmodel(output_dir, deploy_check, self._lstm_app)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 188, in dump_xmodel
    deploy_graphs = get_deploy_graph_list(quantizer.quant_model, quantizer.Nndctgraph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 271, in get_deploy_graph_list
    g_optmizer.constant_folding()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/compile/deploy_optimizer.py", line 83, in constant_folding
    node.set_node_attr(attr_name, attr_val)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_node.py", line 209, in set_node_attr
    self._op.set_attr(key, value)  
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_operator.py", line 272, in set_attr
    self._attrs[attr_name].value = attr_value
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_operator.py", line 126, in value
    f"The type of attr '{self._name.value}' should be {self._type} instead of {type(value[0])}"
TypeError: The type of attr 'kernel' should be <class 'int'> instead of <class 'NoneType'>
```

closed 2022-02-21 by niuxjxlnx