Xilinx/Vitis-AI #556: Adjust CAFFE_CUDA_NUM_THREADS basing on GPU compute capability number

hfkou, 2021-10-15:
How about add the following in ~/include/caffe/util/device_alternate.hpp file, so that it can auto-adjust CAFFE_CUDA_NUM_THREADS according to GPU compute capability number:

// CUDA: use 512/2014 threads per block
// Compute Capability 1.x is 512
// Compute Capability 2.x and above are 1024
#if __CUDA_ARCH__ >= 200
const int CAFFE_CUDA_NUM_THREADS = 1024;
#else
const int CAFFE_CUDA_NUM_THREADS = 512;
#endif

wangxd-xlnx, 2022-02-17:
Hi @hfkou,

Thanks. This is a good suggestion. We will consider adding it into Caffe in the next release.

qianglin-xlnx, 2022-02-22:
HI @hfkou 
Since we haven't received your reply for a long time, I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-22 by qianglin-xlnx

hfkou, 2022-02-22:
Glad if this can be merged into the next release.