Xilinx/Vitis-AI #560: RTX A4000 with vitis ai 1.4

hill19850213, 2021-10-19:
Hi,
I have a problem for NV RTX A4000. This new GPU is not supported cuda 10, and it need to install cuda 11.  For cuda 11, the tensorflow version is 2.4.

Which vitis ai version can support cuda 11 and tensorflow 2.4?

Thanks

hanxue, 2022-03-17:
Hi @hill19850213 @nhphuong91 ,

I have good news for you: the latest Vitis AI v2.0.0 release use based on CUDA 11.0 `nvidia/cuda:11.0-cudnn8-runtime-ubuntu18.04`   https://github.com/Xilinx/Vitis-AI/blob/master/setup/docker/dockerfiles/vitis-ai-gpu.Dockerfile#L1 

Change requests are always welcome. Note though we select the CUDA version that preserves the widest compatibility across all machine learning frameworks, and support as many GPU architectures as possible.

hill19850213, 2022-03-17:
@hanxue ,
I have tried Vitis AI V2.0.0 and there is no cuda problem. Thanks

closed 2022-03-17 by hill19850213