Xilinx/Vitis-AI #577: Meet the " *** Check failure stack trace: *** Aborted ", When run the Vitis-ai demo  on ZCU102

runner42195, 2021-11-02:
When I run the Vitis-ai demo on ZCU102 with the xilinx-zcu102-2020_2 system, I met the bellow issue. I have no idea how to deal with it.

root@xilinx-zcu102-2020_2:~/Vitis-AI/demo/VART/resnet50# ./resnet50 /usr/share/vitis_ai_library/models/resnet50/resnet50.xmodel
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0828 08:09:34.639199  1340 main.cc:285] create running for subgraph: subgraph_conv1
W0828 08:09:44.950841  1340 xrt_cu.cpp:195] cu timeout! device_core_idx 0  handle=0xaaaad76ba6b0 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10010  bo=1 is_done 0 
I0828 08:09:44.950908  1340 xrt_cu.cpp:105] Total: 10010205us	ToDriver: 13us	ToCU: 18446743167732399us	Complete: 0us	Done: 915987344us
F0828 08:09:44.950934  1340 dpu_control_xrt_edge.cpp:177] dpu timeout! core_idx = 0
 LSTART 10  LEND 0  CSTART 0  CEND 0  SSTART 0  SEND 0  MSTART 1  MEND 1  CYCLE_L 3470168485  CYCLE_H 45 
*** Check failure stack trace: ***
Aborted

runner42195, 2021-11-02:
Hi @tianfang-fafafa , can you check it for me?  Thx~

tianfang-fafafa, 2021-11-03:
@runner42195 ，hi，can you please provide md5 check sum of resnet50 resnet50.xmodel and image(xilinx-zcu102-2020_2)?

closed 2021-11-18 by runner42195