Xilinx/Vitis-AI #580: Unsupported op: type: reshape

NeutrinoXY, 2021-11-03:
When trying to compile a neural network that ends with a reshape layer, I got the following output:

```
(vitis-ai-tensorflow2) Vitis-AI /workspace > vai_c_tensorflow2 -m quantized_model_test.h5 -a /opt/vitis_ai/compiler/arch/DPUCVDX8G/VCK190/arch.json -o vai_c_output -n test --options "{'cpu_arch': 'arm64', 'mode': 'normal', 'save_kernel': ''}"
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['quantized_model_test.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/test_org.xmodel', proto=None)
[INFO] tensorflow2 model: quantized_model_test.h5
[INFO] keras version: 2.4.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     : 94%|█████████████████████████████████████▋  | 16/17 [00:00<00:00, 19548.17it/s]               
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==1.4.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/__main__.py", line 49, in main
    runner.normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/runner.py", line 123, in normal_run
    target=target,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/xconverter.py", line 145, in run
    model_files, model_type, _layout, in_shapes, batchsize
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/core.py", line 123, in make_xmodel
    model_type=model_t,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 107, in to_xmodel
    model_type,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 177, in create_xmodel
    name, layers, layout, in_shapes, batchsize
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 2119, in __create_xmodel_from_tf2
    f"*** Unsupported op: type: {op_type}, name: {wrapper_name}"
ValueError: *** Unsupported op: type: reshape, name: tf_op_layer_Reshape
```

The documentation states that the Reshape operation is supported in TF2. Is it because the Reshape is the last layer?

I use Vitis-AI 1.4.1.

zhenzhen-Xilinx, 2022-03-07:
Hello,Sorry for late reply.
Please provide model files.

zhenzhen-Xilinx, 2022-03-07:
Can you share your floating point model and quantized model so we can analyze the problem further. Thank you.

zhenzhen-AMD, 2022-07-20:
Hi @NeutrinoXY ，
This problem has been fixed in Vitis AI 2.5, please use Vitis AI 2.5 docker. Do you have any other questions? If not, we will close this issue. If there is a new issue later, please reopen the issue. Thank you.