Xilinx/Vitis-AI #581: Vitis AI DPU-TRD Vitis Flow resnet fails to classify

Ali-Flt, 2021-11-03:
Hi,
I'm trying to do the [Vitis AI Demo test from vitis platform creation tutorial for zcu104](https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step4.md#test-3-run-a-vitis-ai-demo)

But I have two problems:
First, when I try downloading Vitis AI Library in Vitis IDE I get this error:
[image]

I don't have issues downloading other repositories as Library but Vitis AI git gives me this error, I've even tried many directories for download location but nothing changed.

As a workaround I cloned the Vitis AI git separately an gave its path to Vitis IDE.
Second, by doing this workaround I'm able to build the project completely, but when I try to run the dpu application on ZCU104 board, the resnet model doesn't show correct results for the bell pepper image.

Here are the outputs
```
root@zcu104_custom_plnx:~# env LD_LIBRARY_PATH=samples/lib XLNX_VART_FIRMWARE=/mnt/sd-mmcblk0p1/dpu.xclbin ./dpu_trd b
ellpeppe-994958.JPEG
[ 1748.991559] [drm] Pid 1611 opened device
[ 1748.995524] [drm] Pid 1611 closed device
[ 1748.999547] [drm] Pid 1611 opened device
[ 1749.003469] [drm] Pid 1611 closed device
[ 1749.014716] [drm] Pid 1611 opened device
[ 1749.018667] [drm] Pid 1611 closed device
[ 1749.022609] [drm] Pid 1611 opened device
[ 1749.026530] [drm] Pid 1611 closed device
[ 1749.030646] [drm] Pid 1611 opened device
[ 1749.034660] [drm] Pid 1611 closed device
[ 1749.038987] [drm] Pid 1611 opened device
[ 1749.042912] [drm] Pid 1611 closed device
[ 1749.046862] [drm] Pid 1611 opened device
[ 1749.053860] [drm] zocl_xclbin_read_axlf The XCLBIN already loaded
[ 1749.053870] [drm] zocl_xclbin_read_axlf 1a5daa76-f818-40bb-af8a-c0bee51ee03b ret: 0
[ 1749.064277] [drm] bitstream 1a5daa76-f818-40bb-af8a-c0bee51ee03b locked, ref=1
[ 1749.071954] [drm] Reconfiguration not supported
[ 1749.083714] [drm] bitstream 1a5daa76-f818-40bb-af8a-c0bee51ee03b unlocked, ref=0
[ 1749.083851] [drm] Pid 1611 opened device
[ 1749.095173] [drm] Pid 1611 closed device
[ 1749.099195] [drm] Pid 1611 opened device
[ 1749.103379] [drm] bitstream 1a5daa76-f818-40bb-af8a-c0bee51ee03b locked, ref=1
[ 1749.103400] [drm] Pid 1611 opened device
[ 1749.114536] [drm] Pid 1611 closed device
[ 1749.118473] [drm] Pid 1611 opened device
[ 1749.122508] [drm] bitstream 1a5daa76-f818-40bb-af8a-c0bee51ee03b locked, ref=2
[ 1749.122541] [drm] Pid 1611 opened device
[ 1749.133675] [drm] Pid 1611 closed device
[ 1749.137612] [drm] Pid 1611 opened device
[ 1749.141596] [drm] bitstream 1a5daa76-f818-40bb-af8a-c0bee51ee03b locked, ref=3
[ 1749.142090] [drm] Pid 1611 opened device
[ 1749.153243] [drm] Pid 1611 closed device
[ 1749.157181] [drm] Pid 1611 opened device
score[5]    =  0.00522272   text: electric ray, crampfish, numbfish, torpedo,
score[4]    =  0.00522272   text: hammerhead, hamme[ 1749.161152] [drm] bitstream 1a5daa76-f818-40bb-af8a-c0bee51ee03b locked, ref=4
rhead shark,
score[18]   =  0.00522272   text: magpie,
score[2[ 1750.660487] [drm] bitstream 1a5daa76-f818-40bb-af8a-c0bee51ee03b unlocked, ref=3
0]   =  0.00522272   text: water ouzel, dipper,
score[2]    =  [ 1750.673183] [drm] Pid 1611 closed device
0.00522272   text: great white shark, white shark, man-eater, ma[ 1750.695665] [drm] bitstream 1a5daa76-f818-40bb-af8a-c0bee51ee03b unlocked, ref=2
n-eating shark, Carcharodon carcharias,
[ 1750.695675] [drm] Pid 1611 closed device
[ 1750.877305] [drm] bitstream 1a5daa76-f818-40bb-af8a-c0bee51ee03b unlocked, ref=0
[ 1750.877311] [drm] Pid 1611 closed device
[ 1750.888650] [drm] bitstream 1a5daa76-f818-40bb-af8a-c0bee51ee03b unlocked, ref=0
[ 1750.888653] [drm] Pid 1611 closed device
[ 1750.900065] [drm] Pid 1611 closed device
```

As can be seen the result is incorrect. 
The only change in Tutorial flow I did was the Library repo thing I mentioned above, I'm thinking maybe giving the repo separately to the IDE is not working properly. Is there a fix to this Vitis IDE library download error? Or what else could cause the model to not work properly?

Tested enviroments:
Vitis 2020.2 + Vitis AI branch v1.3
Vitis 2020.2 + Vitis AI branch 1.3.2
Vitis 2021.2 + Vitis AI branch master (1.4)

All environments end in the same result

tianfang-fafafa, 2022-03-04:
hi @Ali-Flt your sdcard image of zcu104-board is generated according to step1-step3 ?

[image]

tianfang-fafafa, 2022-03-08:
hi @Ali-Flt , I follow the step1-step4(test3 Vitis AI Demo),  I can see the corrected result:
[image]

tianfang-fafafa, 2022-03-08:
also, the downloading is available

tianfang-fafafa, 2022-03-08:
but I use Vitis-AI on branch 1.4.    here is dpu info:
root@zcu104_custom_plnx:~# xdputil query
{
    "DPU IP Spec":{
        "DPU Core Count":2,
        "DPU Target Version":"v1.4.1",
        "IP version":"v3.3.0",
        "generation timestamp":"2021-06-07 19-15-00",
        "git commit id":"df4d0c7",
        "git commit time":2106071910,
        "regmap":"1to1 version"
    },
    "VAI Version":{
        "libvart-runner.so":"Xilinx vart-runner Version: 1.4.0-84798c76e6ebb93300bf384cb56397f214676330  2021-08-24-14:04:22 ",
        "libvitis_ai_library-dpu_task.so":"Xilinx vitis_ai_library dpu_task Version: 1.4.0-84798c76e6ebb93300bf384cb56397f214676330  2021-07-27 05:41:52 [UTC] ",
        "libxir.so":"Xilinx xir Version: xir-84798c76e6ebb93300bf384cb56397f214676330 2021-10-14-08:57:31",
        "target_factory":"target-factory.1.4.0 84798c76e6ebb93300bf384cb56397f214676330"
    },
    "kernels":[
        {
            "DPU Arch":"DPUCZDX8G_ISA0_B4096_MAX_BG2",
            "DPU Frequency (MHz)":300,
            "IP Type":"DPU",
            "Load Parallel":2,
            "Load augmentation":"enable",
            "Load minus mean":"disable",
            "Save Parallel":2,
            "XRT Frequency (MHz)":300,
            "cu_addr":"0x80010000",
            "cu_handle":"0xaaaac328b2e0",
            "cu_idx":0,
            "cu_mask":2,
            "cu_name":"DPUCZDX8G:DPUCZDX8G_1",
            "device_id":0,
            "fingerprint":"0x1000020f6014407",
            "name":"DPU Core 0"
        },
        {
            "DPU Arch":"DPUCZDX8G_ISA0_B4096_MAX_BG2",
            "DPU Frequency (MHz)":300,
            "IP Type":"DPU",
            "Load Parallel":2,
            "Load augmentation":"enable",
            "Load minus mean":"disable",
            "Save Parallel":2,
            "XRT Frequency (MHz)":300,
            "cu_addr":"0x80020000",
            "cu_handle":"0xaaaac32997b0",
            "cu_idx":1,
            "cu_mask":1,
            "cu_name":"DPUCZDX8G:DPUCZDX8G_2",
            "device_id":0,
            "fingerprint":"0x1000020f6014407",
            "name":"DPU Core 1"
        }
    ]
}

qianglin-xlnx, 2022-03-16:
Hi @Ali-Flt 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-16 by qianglin-xlnx