Xilinx/Vitis-AI #582: XVDPU-TRD with TFTP boot and NFS rootfs

ksstms, 2021-11-04:
I'm trying to get the XVDPU-TRD 2021.1 version working on a VCK190-ES board with TFTP boot and NFS rootfs.
The 2020.2 version works well, but it seems some things have changed.

I tried to run `test_dpu_runner_mt`, but the whole system froze at this point:
```
root@xilinx-vck190-2021_1:~# test_dpu_runner_mt resnet50-xvdpu-batch-1.xmodel x_0 2
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1104 06:07:35.627738   825 test_dpu_runner_mt.cpp:399] create runner ... 0/2
```

I tried to run the workaround from issue #576, but the system also freezes at the first `devmem` call.

I tried running the TRD from and SD card, and that worked as expected. Then I have modified the PetaLinux settings to use NFS.

One difference I noticed is that in the 2020.2 version the `dpu.xclbin` file is expected to be under `/usr/lib/`. In the new 2021.1 version it's supposed to be under `/media/sd-mmcblk0p1/`. I thought that it might get loaded at boot, so I tried to keep the SD card with the TRD image in the board. Unfortunately, that does not help.

This is my dmesg output before running the test program: [dmesg.txt](https://github.com/Xilinx/Vitis-AI/files/7475091/dmesg.txt)

Does anybody have an idea on what I am missing here? 
Thanks in advance.

qianglin-xlnx, 2021-11-08:
Hi @ksstms 
You can modify the /etc/vart.conf file to specify the new dpi.xclbin path.

ksstms, 2021-11-08:
Thanks for the tip @qianglin-xlnx.
Unfortunately that does not solve the problem.

andrew0917, 2021-11-18:
Hi @ksstms,
     I use the XVDPU-TRD 2021.1 version working on a VCK190-ES board with TFTP boot and NFS rootfs, it works just like SD card.  The result as bleow:
     
[image]

And my TFTP & NFS configurarion as below:
BOOT.BIN--------------storage in SD card
Image,system.dtb,pxelinux.cfg....---------------storage in tftp server directory
rootfs(unpack from rootfs.tar.gz)--------------storage in NFS server directory

ksstms, 2021-11-24:
Thank you @andrew0917 
The difference in my setup is that I'm loading `BOOT.BIN` through JTAG, and `system.dtb` and `pxelinux.cfg/` were missing from my TFTP directory. 
I put the there, and it magically just worked. Then I removed them again, and it still worked.

In the meantime I had to rebuild the project, so it seems like that solved the problem. I have no idea what caused the issue unfortunately.

closed 2021-12-02 by ksstms