Xilinx/Vitis-AI #584: PReLU, Hard-Swish

Nunigan, 2021-11-05:
For the VCK190 it is stated that the activation functions PReLU and Hard-Swish can be used.
Can someone explain how to use this functions for the.
If try to compile it with TF2 it says it is an unsupported op type.

Thanks, Michael

Nunigan, 2021-11-10:
As addition, tf.nn.swish with TF 1.15 works.
tf.keras.layers.Activation('swish') with TF2 does quantize but while compiling for the VCK190 it throws following error.

_* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['quantized_model.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/test_org.xmodel', proto=None)
[INFO] tensorflow2 model: /workspace/test/quantized_model.h5
[INFO] keras version: 2.4.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|██████████████████████████████████████████████████████████████████████████████████████████| 16/16 [00:00<00:00, 21613.16it/s]               
[INFO] infer shape (NHWC)  :100%|██████████████████████████████████████████████████████████████████████████████████████████| 27/27 [00:00<00:00, 36613.71it/s]               
[INFO] perform level-0 opt :100%|██████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 509.42it/s]                   
[INFO] perform level-1 opt :100%|██████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 1432.73it/s]                  
[INFO] infer shape (NHWC)  :100%|██████████████████████████████████████████████████████████████████████████████████████████| 33/33 [00:00<00:00, 47014.96it/s]               
[INFO] generate xmodel     :100%|██████████████████████████████████████████████████████████████████████████████████████████| 33/33 [00:00<00:00, 19652.43it/s]               
[INFO] dump xmodel: /tmp/test_org.xmodel
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCVDX8G_ISA1_C32B1_0601001036088111
[UNILOG][INFO] Graph name: functional_1, with op num: 53
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][XCOM_ACGEN_ERROR][instuction generating fail, please contact us.] 
*** Check failure stack trace: ***
Aborted (core dumped)_

wanghong4compiler, 2022-02-19:
Hi @Nunigan ,
I wonder if your issue has been solved? If not, could you share your floating model, quantization script, dataset and quantized model? Thus we can reproduce the issue and do further analysis. Thank you.

qianglin-xlnx, 2022-03-10:
hi @Nunigan 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-10 by qianglin-xlnx