Xilinx/Vitis-AI #585: C++ Demo questions

zhangchenchen-novauto, 2021-11-09:
`inputs.push_back(std::make_unique<CpuFlatTensorBuffer>(data,inputTensor[0].get()))`

Just like the above code, we talk about binding data and inputs,
data can be int8_t or float32, which is found through comparison,
If the input is int8_t, I need to quantify the input data in advance, float32 does not need,

Excuse me, **if the input is int8_t, must the output be int8_t? Or, the output can be either int8_t or float32?**

yongshengxu-xlnx, 2021-11-10:
No matter what type of your input，the output can be either int8_t or float32. int8_t and float can be converted to each other.
  The input and output of dpu are all int8_t, the float input you give will be converted to int8_t through fixpos and then transmitted to dpu. Similarly, the int8_t output by the dpu will also be converted to float through the output fixpos. 
  If you can directly provide the input of int8_t or directly use the output of int8_t, you don’t need to convert, which will have better performance

zhangchenchen-novauto, 2021-11-10:
@yongshengxu-xlnx 
Just tested it, if the last node of xmodel is fix2float, I set the output to int8_t to receive, it will crash when Run

yongshengxu-xlnx, 2021-11-10:
@zhangchenchen-novauto 
You need to receive data from the last dpu output. check if the size of your outpot is consistent with this layer.
The fix2float node is a cpu node, need to write  code and use the cpu to process it.

zhangchenchen-novauto, 2021-11-10:
@yongshengxu-xlnx
[image]
 
Look, the last node in the above figure is fix2float, and my code is probably written like this: 

`inputs = new int8_t[input_shape];
  outputs = new float[output_shape];
dpurun();
`
I can get the result of the float type, and I feel that there is no problem with using float to connect the data output from the network, **there is no need to write cpu code to convert to float,** but I write like this 
`inputs = new int8_t[input_shape];
  outputs = new int8_t[output_shape];
dpurun();    //It crashed here 
`

yongshengxu-xlnx, 2021-11-10:
```
    batchTensors.push_back(std::shared_ptr<xir::Tensor>(
        xir::Tensor::create(outputTensors[0]->get_name(), out_dims,
                            xir::DataType{xir::DataType::XINT, 8u})));
    outputs.push_back(std::make_unique<CpuFlatTensorBuffer>(
        FCResult, batchTensors.back().get()));
```
Need to specify the output type through DataType, XINT is int8, FLOAT32 is float. When the data type is changed to int8_t, the DateType needs to be changed at the same time

zhangchenchen-novauto, 2021-11-10:
@yongshengxu-xlnx 
OK，我看demo了，多谢 ！

closed 2021-11-10 by zhangchenchen-novauto