Xilinx/Vitis-AI #586: DPU Integration With Vivado: [DNNDK_XRT] Cannot find device, index: 0

Hamptonjc, 2021-11-09:
I followed chapter 5 in the DPU user guide (https://www.xilinx.com/support/documentation/ip_documentation/dpu/v3_3/pg338-dpu.pdf) to integrate the DPU into my block design.

I added the dpu node to the device tree and created the software components with PetaLinux.

However upon using dexplorer I get the error "[DNNDK_XRT] Cannot find device, index: 0"

When trying to run my VART application the dpu controller throws the same error.

`xbutil scan`:
[image]

`cat /proc/interrupts`:
[image]
 
Any ideas?

tianfang-fafafa, 2022-02-17:
@Hamptonjc   hi,  which flow(vitis flow or vivado flow) did you use? and could you please share steps for details?

qianglin-xlnx, 2022-03-10:
Hi @Hamptonjc 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-10 by qianglin-xlnx

Urbont, 2023-03-05:
hi, i think i have met the same problem, could u please share your solutions?

tianfang-fafafa, 2023-03-06:
hi @Urbont, could you please try to add dpu into your block design with v2022.2 tool? the version of dpu match v2022.2 is v4.1. And share your steps to reporpduce issue if issue still exists