Xilinx/Vitis-AI #587: Maths functions

samyhocine, 2021-11-10:
Hi,

On the Xilinx website it's written that there are Math, Linear Algebra and statistics functions included inside the Vitis-AI Library but all I can see is the exponential function (accelerated using SIMD Neon assembly instructions).

I can't see other maths functions (sin/cos...), linear algebra (dot product, cross product, normalisation...), machine learning (no KMeans, no Hierarchical Agglomerative Clustering), statistics...

How is it so ?

ensiehaliagha, 2021-11-10:
Hello, 
I also have the same question. It would be great if they answer and clearly indicate how the acceleration can be done for Math functions.

Hamptonjc, 2021-11-10:
There are other Vitis libraries besides Vitis-AI.

These might be what you are looking for:

https://www.xilinx.com/products/design-tools/vitis/vitis-libraries/vitis-solver.html

https://www.xilinx.com/products/design-tools/vitis/vitis-libraries/vitis-blas.html

samyhocine, 2021-11-15:
Thanks for the feedback ;)

ensiehaliagha, 2021-11-16:
Thanks a lot for providing the links.

closed 2022-03-03 by qianglin-xlnx