Xilinx/Vitis-AI #588: Custom model compilation error on U280

waldenou, 2021-11-10:
Hi,

I am trying to compile a custom model on U280, but I have encountered the following error:

``
/workspace/test/torch_wavenet > vai_c_xir -x quantize_result/WaveNet_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCAHX8H/U280/arch.json -o compiled -n wavenet --options "{'mode':'normal', 'save_kernel':'', 'input_shape': '4,1,1,1024'}"
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][INFO] Target architecture: DPUCAHX8H_ISA2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCAHX8H_ISA2
[UNILOG][INFO] Graph name: WaveNet, with op num: 595
[UNILOG][INFO] Begin to compile...
[UNILOG][WARNING] Operation named WaveNet__WaveNet_959_fix's branch has been removed because they are semantically useless.
[UNILOG][FATAL][XCOM_TILING_FAIL][XCompiler occurs error when tiling. Please contact us.] 
*** Check failure stack trace: ***
Aborted (core dumped)
``
I have attached my model architecture and quantization script below. 

[model_quant1.txt](https://github.com/Xilinx/Vitis-AI/files/7514900/model_quant1.txt)
[model_arch.txt](https://github.com/Xilinx/Vitis-AI/files/7514899/model_arch.txt)

Any help is appreciated! Thanks!

WJ-Xilinx, 2022-03-05:
Hi, @waldenou, could you please share the xmodel(WaveNet_int.xmodel) for a further analysis? 

Thanks!

qianglin-xlnx, 2022-03-28:
Hi @waldenou 
Is this issue solved? If not, could you please share the xmodel to @WJ-Xilinx ? Thanks.

SuZhan0322, 2022-09-01:
Hi @waldenou , please reopen the issue if you still have any other problems. thanks!

closed 2022-09-01 by SuZhan0322