Xilinx/Vitis-AI #589: Vitis AI pytorch quantization problem for yolov5 model

kct890721, 2021-11-12:
Hi everyone,
I'm trying to accelerate a Pytorch YoloV5s model. Here you have the link to the model.
Can someone help to solve this problem,it bothered me for a month.
When quantizing, the following error is arise:
[image]
[image]
[image]

kct890721, 2021-11-12:
here are my cod
[quantize(.py).txt](https://github.com/Xilinx/Vitis-AI/files/7526276/quantize.py.txt)
[common(.py).txt](https://github.com/Xilinx/Vitis-AI/files/7526277/common.py.txt)
[yolo(.py) .txt](https://github.com/Xilinx/Vitis-AI/files/7526278/yolo.py.txt)
e.

carrotandjzy, 2021-11-29:
Can you provide  pytorch version in this case ?

kct890721, 2021-11-29:
hey@ carrotandjzy my pytorch version is  1.10.0+cu102

carrotandjzy, 2021-11-29:
In vitis-1.4.1, torch 1.7 is the highest version supported by vai_q_pytorch.

niuxjxlnx, 2022-02-21:
Pytorch 1.7 version has been supported in Vitis-AI 1.4, and pytorch 1.8, 1.9 version have been supported in Vitis-AI 2.0.
Further more, solution to change pytorch versions among the supported version range is released, please refer to the related part about script replace_pytorch.sh in user guide.

closed 2022-02-21 by niuxjxlnx

krishnaadithya, 2022-03-27:
@kct890721 @niuxjxlnx @carrotandjzy 

Do you guys have the instruction to quantise yolov5 trained model to deploy in kv260?

masrourabdou, 2022-05-31:
have you the problem