Xilinx/Vitis-AI #590: AKS Cmake Error: Could not find a package configuration file provided by "aks"

Ali-Flt, 2021-11-13:
I'm trying to run [AKS examples on edge devices](https://github.com/Xilinx/Vitis-AI/tree/master/tools/AKS#run-examples-on-edge-devices) (ZCU104) but when I run `./cmake-kernels.sh --type=release  --clean` or `./cmake-examples.sh --type=release  --clean` I get errors saying the aks package is not found. I've sourced the sysroot for Vitis-AI so it finds xir and var but no luck with aks.

```
/home/user/Vitis-AI/tools/AKS$ ./cmake-examples.sh --type=release  --clean
-------------------------------------------------------------------
[INFO] Path: /home/user/Vitis-AI/tools/AKS/examples 
-------------------------------------------------------------------
[INFO] /home/user/Vitis-AI/tools/AKS/examples/CMakeLists.txt Found! 
[INFO] Create Build Dir: /home/user/Vitis-AI/tools/AKS/examples/build 
[WRNG] Clean existing configs 
[INFO] Configure Examples ... 
-- Toolchain file defaulted to '/home/user/sysroot/vitis-ai/sysroots/x86_64-petalinux-linux/usr/share/cmake/OEToolchainConfig.cmake'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/user/sysroot/vitis-ai/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/user/sysroot/vitis-ai/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found importable target xir::xir: /home/user/sysroot/vitis-ai/sysroots/cortexa72-cortexa53-xilinx-linux/usr/lib/libxir.so.1.4.1
-- Found importable target unilog::unilog: /home/user/sysroot/vitis-ai/sysroots/cortexa72-cortexa53-xilinx-linux/usr/lib/libunilog.so.1.4.1
CMake Error at CMakeLists.txt:44 (find_package):
  By not providing "Findaks.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "aks", but
  CMake did not find one.

  Could not find a package configuration file provided by "aks" with any of
  the following names:

    aksConfig.cmake
    aks-config.cmake

  Add the installation prefix of "aks" to CMAKE_PREFIX_PATH or set "aks_DIR"
  to a directory containing one of the above files.  If "aks" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/user/Vitis-AI/tools/AKS/examples/build/CMakeFiles/CMakeOutput.log".

[CERR] Configure Failed!
```

Ali-Flt, 2021-11-14:
Here is the files in the aks_edge folder (extracted from aks_edge.tar.gz):

```
tree aks_edge
aks_edge
├── include
│   └── aks
│       ├── AksBatchTensorBuffer.h
│       ├── AksKernelBase.h
│       ├── AksLogger.h
│       ├── AksNodeParams.h
│       ├── AksSysManagerExt.h
│       └── AksTensorBuffer.h
└── lib
    ├── libaks.so -> libaks.so.1
    ├── libaks.so.1 -> libaks.so.1.4.0
    └── libaks.so.1.4.0
```
It seems like the ,cmake files (aks-config.cmake) are missing.

abidrahmank, 2021-11-17:
Hello @Ali-Flt 
You are right, .cmake files are missing. We will correct it in the upcoming release.
Thank you for pointing it out.

VishalX, 2021-11-17:
@Ali-Flt Can you try installing [aks-1.4.1-r77.aarch64.rpm](https://github.com/Xilinx/Vitis-AI/blob/master/tools/AKS/aks-1.4.1-r77.aarch64.rpm) before running and see if it works? 

See the instructions here: https://github.com/Xilinx/Vitis-AI/tree/master/tools/AKS#get-aks-library-kernels-and-examples

hanxue, 2022-03-17:
Hi @Ali-Flt ,

Have you had a chance to try AKS in Vitis AI v2.0 https://github.com/Xilinx/Vitis-AI/tree/v2.0/tools/AKS#get-aks-library-kernels-and-examples ?

qianglin-xlnx, 2022-08-23:
Hi @Ali-Flt 
Closing since no activity for more than 1 month, please reopen if you still have question, thanks

closed 2022-08-23 by qianglin-xlnx