Xilinx/Vitis-AI #591: Is it possible to use DPU with PL memory?

siliangZ, 2021-11-15:
I am new to Vitis AI and VART. I am trying to understand if it is possible to run inference with DPU and PL memory. I searched in the examples. It seems like they all use CpuFlatTensorBuffer to store the input data.(I assume it is PS memory). So I am wondering if there is a API to support user to use DPU with PL memory. And I found that in the execute_async function, it has some copy functions which seems copy data from Host virtual space to physical space. I am wondering if I could just construct the input tensor buffer in the physical space and then I could save the time on copying data. But I don't know how to do that.
And I don't find much document talking about this. Please help me if you know. I would really appreciate it.

wilderfield, 2021-11-15:
VART assumes that the input data to any model is coming from the CPU. Therefore there is always a data transfer from host memory to device memory (ddr). There is also the reverse transfer from device memory (ddr) to host memory at the end of an inference.

Can you please describe the scenario you are trying to support?

deandrew119, 2021-11-15:
From the hardware design, the DPU can only access the PS side DDR though the AXI interface  . In ZU7EV or ZU5EV, only the VCU can add the DDR4 controller at PL side.

siliangZ, 2021-11-15:
@wilderfield Thank you so much for your reply. The scenario is that the input data is generated into PL memory by ADC on PL. So I am trying to avoid coping data around. My assumption is that the device memory is PL DDR. So if we could just give DPU the memory buffer created by ADC, then we could save some time there. Since we don't need to do any other preprocessing on the input data.

siliangZ, 2021-11-15:
@deandrew119 That is good to know. then I think my assumption will not hold any more. I thought DPU has access to PL DDR.

tianfang-fafafa, 2022-02-17:
@siliangZ   hi,  which platform or board （cloud or edge） are you using ？

SlevinZhang, 2022-02-17:
@tianfang-fafafa 
thank you for replying. I am using ultrazed-ev.

qianglin-xlnx, 2022-02-18:
@SlevinZhang 
Can you get the physical address of PL DDR on ARM PS side? 
If yes, you can use "Zero Copy" mode of VART. For "Zero Copy" samples, refer to: https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/VART/vart/dpu-runner/samples/resnet50_zero_copy

tianfang-fafafa, 2022-02-18:
hi @SlevinZhang ,  someone ever did that both CPU and DPU use PL ddr on other board, but performance of PL ddr is low.

qianglin-xlnx, 2022-03-14:
Hi @siliangZ 
Since we haven't received your reply for a long time, I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-14 by qianglin-xlnx