Xilinx/Vitis-AI #592: Is TensorBufferExtImpHostPhy a class we could use to create TensorBuffer on HOST_PHY?

siliangZ, 2021-11-16:
I found that there is a copy_for_input function called if the input buffer is on HOST_VIRT. I would like to avoid doing that copy to speed the process. And I came across the class TensorBufferExtImpHostPhy. I am wondering if that is a valid class we could use to achieve the goal. And if it is, is there a example of using it?

deandrew119, 2021-11-16:
You can refer to "Vitis-AI/tools/Vitis-AI-Library/facedetect/test/test_facedetect_zero_copy.cpp", this sample will show you the interfaces to implement the zero copy. Hope this can meet your requirement.

yongshengxu-xlnx, 2022-02-16:
Hi, @siliangZ 
Can test_facedetect_zero_copy.cpp sample help you achieve your goals? This sample code has the interfaces to implement the zero copy.
Thanks for your feedback. And if there is no feedback, we will close the issue 1 week later.

closed 2022-02-25 by qianglin-xlnx