Xilinx/Vitis-AI #593: LeakyReLU Tensorflow2 multiple subgraphs

Nunigan, 2021-11-16:
I have troubles with LeakyRelu and Tensorflow 2.
If I compile my network with LeakyReLU it generated multiple subgraphs, it looks like it want to deploy the LeakyReLU layer on the CPU.

My dummy network:

_inputs = keras.Input(shape=(256,256,3))
x = Conv2D(64, 3, padding='same')(inputs)
x = LeakyReLU(alpha=0.1)(x)
x = Conv2D(64, 3, padding='same')(x)
x = LeakyReLU(alpha=0.1)(x)_

with _xdputil xmodel test.xmodel -s test.svg_  I get:
[image]

If I am using the normal ReLU:

_inputs = keras.Input(shape=(256,256,3))
x = Conv2D(64, 3, padding='same')(inputs)
x = ReLU()(x)
x = Conv2D(64, 3, padding='same')(x)
x = ReLU()(x)_

 
[image]

I get only one subgraph, as wanted.

With TF1.15 it works fine:
_tf.nn.leaky_relu(x, alpha=0.1)_
 

Thanks for the help, Michael

qberthet, 2021-12-06:
From https://www.xilinx.com/html_docs/xilinx2019_2/vitis_doc/tensorflow_2x.html , it seems that leaky relu is not supported by vai_q_tensorflow2 (but it is by vai_q_tensorflow).

Nunigan, 2021-12-07:
In the UG1414 (v2021.2) November 23, 2021 is stated that LeakyReLU is supported...
Please resolve this misinformation in the documentation.

[image]

qberthet, 2021-12-13:
I've found this document after having answered and I am facing the same issue. I am guessing that it is a matter to use the correct syntax but no success yet..

kylex-xilinx, 2022-03-08:
Hi @qberthet , we cannot reproduce this issue with Vitis-AI 2.0. We checked that LeakyRelu(alpha=0.1) after convolution can be deployed on DPU on our side. Could you please upgrade to 2.0 and try again? If the problem still exists, please share the float toy model for us to reproduce, thanks very much!

qberthet, 2022-03-08:
Yes, I confirm that it seems fixed with Vitis-AI 2.0 for the convolution layers, but I still have issue using LeakyRelu(alpha=0.1) for dense layers : https://support.xilinx.com/s/question/0D52E00006zWXpHSAW/how-to-use-leaky-relu-with-dense-layers-on-vitis-ai-20-tensorflow2?language=en_US . What is the preferred support channel for this kind of issues, Github or Xilinx forums?

qianglin-xlnx, 2022-03-08:
Hi @qberthet 
Please change leaky-relu alpha value from 0.1 to 0.101562.

qberthet, 2022-03-08:
I tried, it does not work, but it does change the behaviour. To provide more information, here are the 3 test cases with some toy models:

**1) LeakyReLU(alpha=0.1) for Conv2D layer, ReLU() for Dense layer:**

Model:
```
inputs = layers.Input(shape=inputs_shape)
x = layers.Conv2D(filters=16, kernel_size=(3, 3))(inputs)
x = layers.LeakyReLU(alpha=0.1)(x)
x = layers.MaxPooling2D(pool_size=(2, 2), strides=(2,2))(x)
x = layers.Conv2D(filters=16, kernel_size=(3, 3))(x)
x = layers.LeakyReLU(alpha=0.1)(x)
x = layers.MaxPooling2D(pool_size=(2, 2), strides=(2,2))(x)
x = layers.Flatten()(x)
x = layers.Dropout(rate=0.5)(x)
x = layers.Dense(units=32)(x)
x = layers.ReLU()(x)
x = layers.Dropout(rate=0.5)(x)
outputs = layers.Dense(units=n_classes, activation='sigmoid')(x)
model = Model(inputs, outputs)
```

VAI compile:
```
fixed_batch_shape: False
WARNING:tensorflow:Unable to restore custom metric. Please ensure that the layer implements `get_config` and `from_config` when saving. In addition, please use the `custom_objects` arg when calling `load_model()`.
[VAI INFO] Update custom_layer_type: []
[VAI INFO] Start CrossLayerEqualization...
10/10 [==============================] - 1s 129ms/step
[VAI INFO] CrossLayerEqualization Done.
[VAI INFO] Start Quantize Calibration...
2022-03-08 14:56:17.744191: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:185] None of the MLIR Optimization Passes are enabled (registered 2)
2022-03-08 14:56:22.440182: I tensorflow/stream_executor/cuda/cuda_dnn.cc:369] Loaded cuDNN version 8201
2022-03-08 14:56:22.871515: I tensorflow/core/platform/default/subprocess.cc:304] Start cannot spawn child process: No such file or directory
2022-03-08 14:56:22.879738: I tensorflow/core/platform/default/subprocess.cc:304] Start cannot spawn child process: No such file or directory
2022-03-08 14:56:22.879789: W tensorflow/stream_executor/gpu/asm_compiler.cc:77] Couldn't get ptxas version string: Internal: Couldn't invoke ptxas --version
2022-03-08 14:56:22.883511: I tensorflow/core/platform/default/subprocess.cc:304] Start cannot spawn child process: No such file or directory
2022-03-08 14:56:22.883635: W tensorflow/stream_executor/gpu/redzone_allocator.cc:314] Internal: Failed to launch ptxas
Relying on driver to perform ptx compilation. 
Modify $PATH to customize ptxas location.
This message will be only logged once.
235/235 [==============================] - 23s 73ms/step
[VAI INFO] Quantize Calibration Done.
[VAI INFO] Start Post-Quantize Adjustment...
[VAI INFO] Post-Quantize Adjustment Done.
[VAI INFO] Quantization Finished.
235/235 [==============================] - 15s 57ms/step - loss: nan - categorical_accuracy: 0.9241
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=[...], model_type='tensorflow2', named_inputs_shape=None, out_filename='...', proto=None)
[INFO] tensorflow2 model: ...
[INFO] keras version: 2.6.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 13/13 [00:00<00:00, 12146.57it/s]               
[INFO] infer shape (NHWC)  :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 24/24 [00:00<00:00, 10731.69it/s]               
[INFO] perform level-0 opt :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 349.64it/s]                   
[INFO] perform level-1 opt :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 1191.23it/s]                  
[INFO] infer shape (NHWC)  :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 26/26 [00:00<00:00, 5180.37it/s]                
[INFO] generate xmodel     :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 26/26 [00:00<00:00, 1801.89it/s]                
[INFO] dump xmodel: ...
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B1152_01000020F6012203
[UNILOG][INFO] Graph name: model, with op num: 38
[UNILOG][INFO] Begin to compile...
[UNILOG][WARNING] xir::Op{name = quant_dense_1_sigmoid_sigmoid(MoveReshape)(MergeMultoHsigmoid), type = hard-sigmoid-fix} has been assigned to CPU: [This target does not support single hard-sigmoid.].
[UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
[UNILOG][INFO] Compile done.
[UNILOG][INFO] The meta json is saved to "..."
[... 144 lines cut ...]
[INFO] generate xmodel     :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 27/27 [00:00<00:00, 1124.49it/s]                
[INFO] dump xmodel: ...
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B1152_01000020F6012203
[UNILOG][INFO] Graph name: model, with op num: 39
[UNILOG][INFO] Begin to compile...
[... cut ...]
[UNILOG][INFO] The compiled xmodel's md5sum is 7d0b76227d3b9ab3029936f23f987865, and has been saved to ...
```

Resulting graph:

[image]

So, now the compiler complain that
```LeakyReLU layer leaky_re_lu_2(alpha=0.1015620008111) is not supported by DPU, currently DPU only supports LeakyReLU layer with `alpha=0.1`.```
(Which it did not in case 2) and the LeakyReLU is still performed on the CPU (but the graph did change).

Would this be helpfull to pinpoint where I am doing something wrong, or if there is a bug in VAI2.0 ?
Thanks in advance!

wanghong4compiler, 2022-03-21:
Hi @Nunigan ,

Thank you very much for your feedback.
About the issue that matmul with leaky-relu partitioned to CPU, it's a bug of compiler, it would be fixed in next release, VAI2.5.
About the alpha number, it's a print issue, actually, it should be float (26/256). Also, it would be fixed in next release.
So stay tuned, and thank you again.

DenverConger, 2022-05-17:
Hello @wanghong4compiler is there any update to the LeakyRelu issues? I would really really love to be able to use it in my current Vitis AI project! thanks!

ThomasGoud, 2022-07-07:
Hello, I encounter similar issue when compiling Yolov3 with TF2 on ZCU104.

- The layers in the model are the following: CONV2D-->BATCH_NORM-->LEAKY RELU
- I'm using alpha=0.1 for LeakyRelu which is converted to 26/256 (confirmed in netron) during  quantization
- As it can be seen in the resulting graph, the compiler divide each leakyRelu in subgraph for cpu computation:
[image]

Do you have any solution for computing everything in DPU and not CPU? In my case, the LeakyRelu is after a Batch_Normalization which, I suppose, prevent the fusion of LeakyRelu inside the Conv layer.

Thanks in advance,
Thomas

wanghong4compiler, 2022-07-07:
Hi @DenverConger ,
have you tried the newest release VAI 2.5?

wanghong4compiler, 2022-07-07:
Hi @ThomasGoud ,
the alpha should be 0.1015625, it can't be round to 0.1, could you try this again?

closed 2022-07-11 by qianglin-xlnx