Xilinx/Vitis-AI #596: [Question]Does Vitis AI support ONNX transformed models?

laumecha, 2021-11-18:
I have an ONNX model and I want to quantize the model using Tensorflow. Then, I want to use the quantized model in another framework that requires the ONNX format. The flow that I want to do is the following:
First, I want to import an ONNX model into TensorFlow. In [this](https://github.com/onnx/tutorials/blob/master/tutorials/OnnxTensorflowImport.ipynb) tutorial, I saw the way to do it.
Then, I want to execute the model quantization with the Vitis AI tools or directly Tensorflow and, as the last step, I want to export the model to ONNX format using tf2onnx, like in [this](https://github.com/onnx/tensorflow-onnx/blob/master/tutorials/keras-resnet50.ipynb) example.

[Here](https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_4/ug1414-vitis-ai.pdf) says that there is an ONNXruntime for Vitis AI (which I assume is deprecated as the files do not correspond to the actual Vitis AI GitHub state). However, page 44 says: "The quantized model cannot be taken in by the standard version of TensorFlow, PyTorch, or Caffe framework.". So, my questions are:

1. Should I use directly TensorFlow for the incompatibility with the Vitis AI models?
2. Can be used any Tensorflow model by Vitis AI?
3. Why the quantized model cannot be taken in by the standard version?
4. If the quantized model cannot be taken in by the standard version, why there is ONNX model export support integrated into Vitis AI?

xieyi4650, 2022-07-22:
Hi @laumecha 
1. what version of TensorFlow are you using, 1.x or 2.x?
2. yes, Vitis Ai can use any model generated by standard TensorFlow
3. there is quantize op in the quantized model.  For TensorFlow, this is a custom OP, so cannot be  taken directly, if you want to take,  load the custom op
4. we load it

qianglin-xlnx, 2022-08-23:
Hi @laumecha 
Closing since no activity for more than 1 month, please reopen if you still have question, thanks

closed 2022-08-23 by qianglin-xlnx