Xilinx/Vitis-AI #597: How to get video output file from the main_for_video_demo function of demo.hpp?

Ali-Flt, 2021-11-21:
I'm trying to develop an application that uses Vitis AI Library models. Using the demo functions such as main_for_video_demo seem to be much easier that writing my own code because multi-threading is handled in that code very good. 
But Instead of a display output, I want to write the output to a video file. Is there a way to do this with the same function?

I tired changing line 36 of the demo.hpp file from
`DEF_ENV_PARAM(DEMO_USE_VIDEO_WRITER, "0")` to `DEF_ENV_PARAM(DEMO_USE_VIDEO_WRITER, "1")`

but running the app just gives me this output below. It looks like the gst pipe can't be created for some reason.
Also I think there are some plugins missing from the gst package of petalinux because even with having opencv and petalinux package groups included in the rootfs, I can't give a MP4 file as input to to main_for_video_demo function so please look into that too.

```
[ WARN:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp (935) open OpenCV | GStreamer warning: Cannot query video position: status=1, value=5, duration=407
[ WARN:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp (1760) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module kmssink0 reported: Could not open DRM module xlnx
[ WARN:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp (1760) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module kmssink0 reported: GStreamer error: state change failed and
some element failed to post a proper error message with the reason for the failure.
[ WARN:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp (1592) open OpenCV | GStreamer warning: GStreamer: cannot put pipeline to play

WARNING: Logging before InitGoogleLogging() is written to STDERR
F1121 19:17:13.843691  1412 demo.hpp:386] cannot open gst: appsrc ! videoconvert ! queue ! kmssink driver-name=xlnx plane-id=39 fullscreen-overlay=false sync=false
*** Check failure stack trace: ***
Aborted
```

ajiang17, 2022-01-12:
I met the same error。 Have you solve it ??

zz002, 2022-02-21:
You need to use environment variables to specify save video information to a file and specify related file information
`env DEMO_USE_VIDEO_WRITER=1 DEMO_VIDEO_WRITER="appsrc ! videoconvert ! queue ! filesink location=/home/root/test.mp4"`

closed 2022-03-02 by qianglin-xlnx