Xilinx/Vitis-AI #598: HardSwish HardSigmoid are not supported in Pytorch Quantization?

chuong98, 2021-11-23:
Hello, I am compiling a model with HardSwish/HardSigmoid.

I can run the `calib` mode, and the model are successfully calibrated. In the output `model.py` that is generated automatically, I can see that:
```python
def __init__(self):
      self.module_133 = py_nndct.nn.Conv2d(in_channels=192, out_channels=192, kernel_size=[1, 1], stride=[1, 1], padding=[0, 0], dilation=[1, 1], groups=1, bias=True)
      self.module_134 = py_nndct.nn.Hardswish(inplace=True) 
def forward(self, *args):
      self.output_module_134 = self.module_134(self.output_module_133)
```
but when I run the `deploy` mode to compile the `_int.xmodel`, it generates error:
```
File "/.../lib/python3.6/site-packages/nndct_shared/compile/xop_creator.py", line 649, in hswish
    mul_fp[0] = quant_config['output'][node.name][0] 
KeyError: 'Model/ConvModule[short_conv]/Hardswish[activate]/11363'
During handling of the above exception, another exception occurred:
  File ".../lib/python3.6/site-packages/nndct_shared/compile/xir_compiler.py", line 136, in do_compile
    raise AddXopError(node.name, node.op.type, str(e)
nndct_shared.utils.exception.AddXopError: Failed to add op(name:Model/CSPLayer[1]/ConvModule[short_conv]/Hardswish[activate]/11363, type:hswish) in xGraph.
```
How should I fix the error ?

Alexiazzf, 2021-11-30:
Hello, I tested the model you showed in the Docker of xilinx/vitis-ai:1.4.1.978(gpu). It gose well when I run the deploy mode to compile the _int.xmodel. Have you checked quant_info.json to see if they all have fix points? Or you can show more details about your model.

Alexiazzf, 2021-11-30:
> Hello, I am compiling a model with HardSwish/HardSigmoid.
> 
> I can run the `calib` mode, and the model are successfully calibrated/evaluated. In the output `model.py` that is generated automatically, I can see that:
> 
> ```python
> def __init__(self):
>       self.module_133 = py_nndct.nn.Conv2d(in_channels=192, out_channels=192, kernel_size=[1, 1], stride=[1, 1], padding=[0, 0], dilation=[1, 1], groups=1, bias=True)
>       self.module_134 = py_nndct.nn.Hardswish(inplace=True) 
> def forward(self, *args):
>       self.output_module_134 = self.module_134(self.output_module_133)
> ```
> 
> but when I run the `deploy` mode to compile the `_int.xmodel`, it generates error:
> 
> ```
> File "/.../lib/python3.6/site-packages/nndct_shared/compile/xop_creator.py", line 649, in hswish
>     mul_fp[0] = quant_config['output'][node.name][0] 
> KeyError: 'Model/ConvModule[short_conv]/Hardswish[activate]/11363'
> During handling of the above exception, another exception occurred:
>   File ".../lib/python3.6/site-packages/nndct_shared/compile/xir_compiler.py", line 136, in do_compile
>     raise AddXopError(node.name, node.op.type, str(e)
> nndct_shared.utils.exception.AddXopError: Failed to add op(name:Model/CSPLayer[1]/ConvModule[short_conv]/Hardswish[activate]/11363, type:hswish) in xGraph.
> ```
> 
> I am using the Vitis-AI 1.4 with Docker. How should I fix the error ?

Hello, I tested the model you showed in the Docker of xilinx/vitis-ai:1.4.1.978(gpu). It gose well when I run the deploy mode to compile the _int.xmodel. Have you checked quant_info.json to see if they all have fix points? Or you can show more details about your model.

Alexiazzf, 2021-11-30:
Hi @chuong98 ,I tested the model you showed in the Docker of xilinx/vitis-ai:1.4.1.978(gpu). It gose well when I run the deploy mode to compile the _int.xmodel. Have you checked quant_info.json to see if they all have fix points? Or you can show more details about your model.

chuong98, 2021-12-02:
Hi, thank you. I make a minimal model with only one [conv-act], 
and the model can be compiled with HardSwish and HardSigmoid.
Unfortunately, these activation can only run on CPU.

This is the minimal code:
```python
import torch 
import torch.nn as nn
from pytorch_nndct.apis import torch_quantizer
import argparse

class Model(nn.Module):
	def __init__(self):
		super(Model, self).__init__()
		self.conv = nn.Conv2d(3, 8, 3, 1, 1, bias=False)
		# self.act = nn.Hardswish(inplace=True)
		self.act = nn.Hardsigmoid()

	def forward(self, x):
		x = self.conv(x)
		x = self.act(x)
		return x

def parse_args():
	parser = argparse.ArgumentParser(description='mmcls test model')
	parser.add_argument('--out-dir', 
						default='cache/test_hswish/quantized',
						help='output result folder')
	parser.add_argument('--quant-mode', 
						default='calib', 
						choices=['calib', 'test'], 
						help='quantization mode. float: no quantization, evaluate float model, calib: quantize, test: evaluate quantized model')
	parser.add_argument('--deploy', 
						dest='deploy',
						action='store_true',
						help='export xmodel for deployment')
	args = parser.parse_args()
	return args

  
if __name__ == '__main__':
	args = parse_args()
	model = Model()
	model.eval()
	device = torch.device("cuda" if torch.cuda.is_available() else "cpu")
	quant_mode = args.quant_mode

	# Quantize model only needs a dummy input
	input = torch.randn([32, 3, 32, 32])
	quantizer = torch_quantizer(args.quant_mode, model, input, 
					output_dir=args.out_dir, device=device)
	quant_model = quantizer.quant_model

	if args.deploy:
		#	To generate the xmodel for compilation, the batch size should be 1. 
		assert quant_mode == 'test', 'Only test mode can generate xmodel'
		input = torch.randn([1, 3, 32, 32])
	

	# Run model on dummy input
	input = input.to(device)
	quant_model(input)

	if quant_mode == 'calib':
		quantizer.export_quant_config()
	if args.deploy:
		quantizer.export_xmodel(output_dir=args.out_dir, deploy_check=False)
```

chuong98, 2021-12-02:
Also, when I tried with ReLU6, it warns:
`[VAIQ_WARN]: ReLU6 has been replaced by ReLU.` 
Does that mean ReLU6 is not really supported?

Alexiazzf, 2021-12-06:
Hi @chuong98, 

I tested your mini model in the xilinx/vitis-ai:1.4.1.978 docker. It can be quantized and deployed with cuda. However, only VCK190 support HardSwish and HardSigmoid compiling. Other hardware will transfer HardSwish and HardSigmoid to CPU automatically.

Besides, we do support ReLU6. The warning you have got is for equalization to improve quantization accuracy.

closed 2021-12-07 by chuong98

white0dew, 2022-01-18:
> Hi @chuong98,
>
> I tested your mini model in the xilinx/vitis-ai:1.4.1.978 docker. It can be quantized and deployed with cuda. However, only VCK190 support HardSwish and HardSigmoid compiling. Other [...]
could you explain how to deploy HardSwish on CPU？ is there any example？
Thanks