Xilinx/Vitis-AI #600: How to process outputs from DPU for my custom model

wanghao16020510036, 2021-11-26:
I see the process of the host program is generally like this:
```C++
//create runner
auto runner = vart::Runner::create_runner(subgraph[0], "run");

//get  some infomation of inputTensors and outputTensors
auto inputTensors = runner->get_input_tensors();
auto outputTensors = runner->get_output_tensors();
int inputCnt = inputTensors.size();
int outputCnt = outputTensors.size();
TensorShape inshapes[inputCnt];
TensorShape outshapes[outputCnt];
shapes.inTensorList = inshapes;
shapes.outTensorList = outshapes;
getTensorShape(runner.get(), &shapes, inputCnt, outputCnt);
auto out_dims = outputTensors[0]->get_shape();
auto in_dims = inputTensors[0]->get_shape();
auto input_scale = get_input_scale(inputTensors[0]);
auto output_scale = get_output_scale(outputTensors[0]);
int outSize = shapes.outTensorList[0].size;
int inSize = shapes.inTensorList[0].size;
int batchSize = in_dims[0];

//Allocate memory for input and output(s)
int8_t* imageInputs = new int8_t[inSize * batchSize];
int8_t* result = new int8_t[batchSize * outSize];

//read input data into imageInputs

std::vector<std::unique_ptr<vart::TensorBuffer>> inputs, outputs;
std::vector<vart::TensorBuffer*> inputsPtr, outputsPtr;
std::vector<std::shared_ptr<xir::Tensor>> batchTensors;
batchTensors.push_back(std::shared_ptr<xir::Tensor>(
        xir::Tensor::create(inputTensors[0]->get_name(), in_dims,
                            xir::DataType{xir::DataType::XINT, 8u})));
inputs.push_back(std::make_unique<CpuFlatTensorBuffer>(imageInputs, batchTensors.back().get()));
batchTensors.push_back(std::shared_ptr<xir::Tensor>(
        xir::Tensor::create(outputTensors[0]->get_name(), out_dims,
                            xir::DataType{xir::DataType::XINT, 8u})));
outputs.push_back(std::make_unique<CpuFlatTensorBuffer>(result, batchTensors.back().get()));
inputsPtr.push_back(inputs[0].get());
outputsPtr.push_back(outputs[0].get());
auto job_id = runner->execute_async(inputsPtr, outputsPtr);

//run
runner->wait(job_id.first, -1);
```
For example, if the output tensor shape of my model in pytorch is torch.size([10, 10, 3]), then this tensor will be flattened into a 1-dimensional array with with a length of 300. I want to know that in this host app, how to get the element of  out[3][4][2] in pytorch.

wanghao16020510036, 2021-11-26:
Is there an API for indexing in vart？ @hanxue @eemars @stsoe @woinck

yongshengxu-xlnx, 2022-02-16:
Hi, @wanghao16020510036 
These is no API for indexing, indexing needs to be done by yourself. 
To get the element of out[3][4][2], you can try:
```
auto idx = 3*width*c + 4*c+2; // Be like: 3*10*3+4*10+2 = 132;
auto element = results[idx];
```
Hope these help you.

qianglin-xlnx, 2022-02-25:
Hi @wanghao16020510036 
Has this issue been solved?

wanghao16020510036, 2022-03-01:
> Hi @wanghao16020510036 Has this issue been solved?

After looking at a demo carefully I understand how to code, as mentioned above. But I received your reply more than three months after I raised this question. Your reply was too late!

qianglin-xlnx, 2022-03-01:
@wanghao16020510036 
Glad to hear you have solved this issue, and sorry for the late reply.

closed 2022-03-01 by qianglin-xlnx