Xilinx/Vitis-AI #603: Non-consistent quantization results

RaffaeleBerzoini, 2021-11-30:
Hi, i'm using VAI 1.4.1.978.

I'm training and quantizing a 16 Million parameters UNET (tensorflow 2.3.0)

I saved the model at different epochs and quantize it with the same calibration set. To explain my problem i'll report my f1 scores:
-Epoch 22: float_model 89.9% and quant_model: 89.1%
-Epoch 29: float_model 90.1% and quant_model: 77.0%
-Epoch 37: float_model 90.8% and quant_model: 86.2%
-Epoch 43: float_model 90.9% and quant_model: 88.3%
-Epoch 55: float_model 91.2% and quant_model: 69.3%

All the evaluation are made on the same split of the dataset and during training (at least till epoch 60) there no overfitting is happening.

What can be the problem? I've faced the same problem also with the same Unet but with more layers (~23M parameters). Is there anything i can do to have more homogeneous results?

I'm not scaling the data (input have data type uint8) and i'm using 300 images for the calibration dataset (results does not change with different calibration dataset sizes or by using fast finetuning with different amount of images and fast_ft_epochs=50).

Thanks

xieyi4650, 2022-03-07:
Hi @RaffaeleBerzoini 
Accuracy improvements in floating-point models do not necessarily lead to improved accuracy in quantized models.
It is possible to make the distribution of weights difficult to quantize during the training process.
you can try to quantize aware training model after float training which generate a good float model with high f1 scores.

Can provide your model or build model code for us to analyze.

RaffaeleBerzoini, 2022-03-07:
Hi @xieyi4650 
Thank you for your reply. I actually found what the problem was: 
I was training with uint8 images for memory concerns. Then i decided to try training my model with float images scaled between -1 and 1 (same preprocessing and nothing else changed except for the values range). Now the quantization step works perfectly with negligible performance loss.

I'm not sure why uint8 input images led to this problem but this seemed to be the solution. 

Thank you again