Xilinx/Vitis-AI #605: vai_q_pytorch : Export Quant Config KeyError

schegde, 2021-12-02:
Hi,
   When I try to quantize a particular layer of a model , having layer structure:
```
   nn.Sequential(
            nn.Conv2d(192, 64, 1, bias=False, padding=0),
            nn.BatchNorm2d(64),
            nn.ReLU(),
            )
```
  I get through fast finetune step, generating the `bias_corr.pth`, `param.pth` and the <model>.py file, but the `export_quant_config()` step errors out in a particular `base_quantizer.py` file of the `vai_q_pytorch` module, saying the particular output tensor key not found in the `self._QuantInfo` variable as shown below:
[image]
Any pointers on how to go about investigating/resolving this issue further would be appreciated! Please let me know what information is needed to help with this further. I am using pytorch v1.9.0 version, and the quantizer tool is installed from source in my environment.

MarcNuFu, 2021-12-09:
Hi,
I'm facing the same issue. Any update on how to fix it?

PeterQuinn396, 2022-01-14:
Hi,

I am also facing a similar issue. 

Running the Vitis-AI latest docker on WSL2 Ubuntu 20.04 with the `vitis-ai-pytorch` conda environment.

When I call ` quantizer.export_quant_config()`, I get the following error:

```python
Traceback (most recent call last):
  File "quantize.py", line 88, in <module>
    main()
  File "quantize.py", line 84, in main
    quantize(model, args.quant_mode, batch_size=1)
  File "quantize.py", line 64, in quantize
    quantizer.export_quant_config()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 94, in export_quant_config
    self.processor.export_quant_config()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 161, in export_quant_config
    self.quantizer.export_quant_config()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/quantization/torchquantizer.py", line 401, in export_quant_config
    self.organize_quant_pos()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/quantization/torchquantizer.py", line 346, in organize_quant_pos
    bnfp = self.get_bnfp(src_name, False)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/quantization/base_quantizer.py", line 117, in get_bnfp
    bnfp = copy.deepcopy(self._QuantInfo[tensor_type][name])
KeyError: 'HandGestureNet::HandGestureNet/809'
```

`

niuxjxlnx, 2022-01-20:
Suppose you guys are using quant_stub/dequant_stub OP to do partial quantization on a network. There maybe some problem about quant_stub OP is needed by not inserted somewhere in the network. You can try vitis-ai 2.0, which will come next week. 
In Vitis-ai 2.0, this kind of problem will be detected automatically and reported.

PeterQuinn396, 2022-01-23:
I've just tested out vitis-ai 2.0. My issue seems to have been solved. No more errors and I get a `quant_info.json` output. 

@niuxjxlnx Thank you very much for checking this and providing some advice!

closed 2022-02-21 by niuxjxlnx