Xilinx/Vitis-AI #607: Error response from daemon: No such image: xilinx/vitis-ai-gpu:1.4.1.978

JasonTryharder, 2021-12-04:
I have following [this](https://github.com/Xilinx/Vitis-AI) to install vitisAI and trying to install gpu docker, 
below is mine configuration : 
Linux C001886040 5.11.0-41-generic#45~20.04.1-Ubuntu SMP Wed Nov 10 10:20:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
Nvidia RTX A3000,  NVIDIA-SMI 495.44       Driver Version: 495.44       CUDA Version: 11.5
Vitis unified platform 2021.1 
Docker version 20.10.11, build dea9396
cmake version 3.20.2

When run ./docker_build_gpu.sh 
it shows : 
The following information may help to resolve the situation:
[Vitis-AI_docker_gpu_build_Error.txt](https://github.com/Xilinx/Vitis-AI/files/7654772/Vitis-AI_docker_gpu_build_Error.txt)

The following packages have unmet dependencies:
 xrt : Depends: libboost-filesystem1.71.0 but it is not installable
       Depends: libboost-program-options1.71.0 but it is not installable
       Depends: libc6 (>= 2.29) but 2.27-3ubuntu1.4 is to be installed
       Depends: libncurses6 (>= 6) but it is not installable
       Depends: libprotobuf17 but it is not installable
       Depends: libtinfo6 (>= 6) but it is not installable
E: Unable to correct problems, you have held broken packages.
The command '/bin/bash -c wget --progress=dot:mega -O xrt.deb ${XRT_URL}     && ls -lhd ./xrt.deb     && apt-get update -y     && apt-get install -y ./xrt.deb     && rm -fr /tmp/*' returned a non-zero code: 100
Error response from daemon: No such image: xilinx/vitis-ai-gpu:1.4.1.978
[image]

Thanks

JasonTryharder, 2021-12-04:
I have found the same issue when creating cpu docker ./docker_build_cpu.sh
[image]
[image]

full error msg: 
[Vitis-AI_docker_cpu_build_Error.txt](https://github.com/Xilinx/Vitis-AI/files/7654798/Vitis-AI_docker_cpu_build_Error.txt)

schegde, 2021-12-06:
Hi,  I am also encountering similar issue, but different error message, when I try to build the `docker_build_gpu.sh` . I am on Ubuntu 18.04 host PC.
[image]

nhphuong91, 2021-12-09:
@JasonTryharder vitis-ai hasn't supported Nvidia Ampere generation (yours is A3000). To be concrete, vitis-ai hasn't supported CUDA 11 + CUDA 10 don't support Nvidia Ampere gen
About building vitis-ai cpu, you can pull the pre-built docker image from here: https://hub.docker.com/r/xilinx/vitis-ai-cpu/tags
@schegde Try to switch to tag 'v1.4' or 'v1.4.1'. I'm usually avoid using master branch, it's inconsistent

hanxue, 2022-03-17:
@JasonTryharder @schegde ,

Are you guys still having error message building the Vitis AI GPU docker image? 

Odd to see package conflict when installing XRT dependencies - boost, libc and so on. XRT uses the OS' default package version, and for Vitis AI GPU it is based on Ubuntu 18.04.

siniGAL, 2022-03-17:
I have some problem:
```bash
make[2]: Leaving directory '/tmp/protobuf-3.4.0/src'
Makefile:1488: recipe for target 'all-recursive' failed  
make[1]: Leaving directory '/tmp/protobuf-3.4.0'
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
Makefile:1395: recipe for target 'all' failed
The command '/bin/bash -c cd /tmp; wget --progress=dot:mega https://codeload.github.com/google/protobuf/zip/v3.4.0     && unzip v3.4.0     && cd protobuf-3.4.0     && ./autogen.sh     && ./configure     && make -j     && make install     && ldconfig     && rm -fr /tmp/*' returned a non-zero code: 2  
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103  
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103
```

hanxue, 2022-03-17:
@siniGAL Please raise a separate Github Issue. These are different build errors. 

Are you building from the latest v2.0 release?

I see multiple lines of 

```
[0m[91mg++: internal compiler error: Killed (program cc1plus)
```

This may happen due to lack of resources on the docker build host. How much memory is allocated for docker build, and is there sufficient free disk space - recommended 100GB?

siniGAL, 2022-03-18:
@hanxue thanks.
I create a new issue. I solved it.

closed 2022-04-02 by qianglin-xlnx

mykishan, 2022-06-10:
[image]
Getting error while executing 

> ./docker_build_cpu.sh

returned a non-zero code: 137
Error response from daemon: No such image: xilinx/vitis-ai-cpu:2.0.0.1103
Error response from daemon: No such image: xilinx/vitis-ai-cpu:2.0.0.1103
Error response from daemon: No such image: xilinx/vitis-ai-cpu:2.0.0.1103