Xilinx/Vitis-AI #610: opt/vitis_ai/compiler/arch.json

Marc-lu, 2021-12-08:
After I quantized the resnet18_quant.py, I compile the xmodel generated by pytorch quantizer using cmd like "vai_c_xir -a /opt/vitis-ai/compiler/arch/arch.json -x ./quantize_result/ResNet_int.xmodel -o ./compile_result/ -n resnet18" and encount the error that "No file or directory opt/vitis_ai/compiler/arch.json "
I look for  arch.json in  /opt/vitis-ai/compiler/arch/ while I find there is even no path like opt/vitis_ai/...  
I want to generate the arch.json by myself and the json file is like {"target":"dpuv2","cpu_arch":"arm64"}. There comes another error like "ERROR: NO FRONT END SPECIFIED" I think the arch.json file may lack some item like "dcf":"" but I didn't find any related file like zcu104_....json about my evaluate board(ZCU104) or any file as the format"xxx.dcf"
So I want to ask for help about how to find or generate the file arch.json.

qianglin-xlnx, 2021-12-08:
Hi @Marc-lu 
Find the ZCU104 arch.json file in /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json 

Vitis-AI /workspace > ls /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json
/opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json
Vitis-AI /workspace > cat /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json
{
    "target": "DPUCZDX8G_ISA0_B4096_MAX_BG2"
}

And refer to https://github.com/Xilinx/Vitis-AI/blob/master/dsa/DPU-TRD/prj/Vitis/README.md#522-get-json-file to get the arch.json

Marc-lu, 2021-12-08:
> Hi @Marc-lu Find the ZCU104 arch.json file in /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json
>
> Vitis-AI /workspace > ls /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json [...]

Thanks a lot. It works now. I thought the path existed on ubuntu os before.

closed 2021-12-24 by qianglin-xlnx