Xilinx/Vitis-AI #612: Based on VCK5000, error:-19 failed to fully probe device，and can't load firmare,give up.

China-CC-lon, 2021-12-09:
[image]
I tested different scripts and source code compilation in two different environments based on Ubuntu 20.04.2-based kernel 5.4.0-74 and Ubuntu 18.04.5 kernel 5.4.0-74, including the ./install script on GitHub running vck5000 And the ./install script in the release file downloaded from the file sent by the security station center, the source code compilation includes the xrt<version>.deb file, but even though I have tried so many installation situations, it always reaches 100% during the installation process The above problem occurred at 96 o'clock, and it always said that my firmware could not be loaded. Why is this? Can you tell me the solution to this problem?
(I connected the VCK5000 through the PCIE interface of the graphics card, and I can detect that the board is connected and existing)

ombhilare999, 2021-12-29:
We are also facing same issue on our vck5000 board, have you got any working solution for this issue @China-CC-lon ?

China-CC-lon, 2022-01-07:
I don’t know, and I’ve been facing this problem all the time. I’ve tried many methods and the firmware can’t be loaded. Some people say that the kernel does not match, and some people say that the computer’s CPU does not work. The strange thing is that many people directly compile the script and install it directly. , And I always have the same problem when compiling scripts or compiling source code, so I ask questions here.

ombhilare999, 2022-01-07:
https://gist.github.com/ombhilare999/130a920da6d9a90fa0a833da059acec3
One of the Xilinx engineers told us thst shell is not matching, that's why the best way is to flash new pdi on to the board. I have compiled all the steps on the above link.
If you are trying that on your board then let me know the results.

Regards,
Omkar Bhilare
ombhilare999@gmail.com

China-CC-lon, 2022-01-19:
It was fine after I replaced a CPU and motherboard.

closed 2022-01-19 by China-CC-lon