Xilinx/Vitis-AI #613: AddXopError: Supporting custom ops in xGraph

schegde, 2021-12-14:
Hi, 
  I am able to get the model finetuned to an accuracy close to the floating point accuracy, but when I go for the `deploy` option to dump the `xmodel`, I get following exception: 
```
[VAIQ_NOTE]:` =>Converting to xmodel ...
[VAIQ_ERROR]: Failed convert graph to xmodel.
nndct_shared.utils.exception.AddXopError: Please support ops 'aten::type_as', in xGraph.
```
Is there any documentation available on how to add custom ops support to xGraph? Any pointers on how to get started on that would be appreciated!

bryanloz-xilinx, 2022-01-20:
@fanz-xlnx ?

fanz-xlnx, 2022-01-20:
I hope the following link work for you
https://github.com/Xilinx/Vitis-AI/tree/master/demo/Custom_OP_Demo/op_add

schegde, 2022-01-25:
Thank you so much! I see that the custom OP flow was newly introduced in Vitis AI 2.0! I had created this issue with VAI 1.4 release. Let me try the custom OP flow, and so will close this issue for now.

closed 2022-01-25 by schegde