Xilinx/Vitis-AI #614: How can I get the temperature of VCK5000 using the SMBus/I2C interface

whyaysd, 2021-12-16:
I found the SC document here: https://xilinx.github.io/Alveo-Cards/master/management-specification/oob-intro.html

And It conveys that we can get the temperature of my card using this interface.
However, I can not find the I2C address through i2cdetect command.

How can I using this Interface to get those information?

hanxue, 2022-03-17:
Does this XRT command give you the information that you need?

```
sudo /opt/xilinx/xrt/bin/xbutil --legacy dump  -d 0000:d8:00.0
```

SuZhan0322, 2022-09-06:
Hi @whyaysd ,
In VAI2.5, you can get the temperature of VCK5000 using XRT command:
/opt/xilinx/xrt/bin/xbutil examine --report thermal --device
thanks!

SuZhan0322, 2022-09-29:
Hi @whyaysd ,
Any update about this issue? Thanks!

SuZhan0322, 2022-10-14:
Hi @whyaysd ,
Please reopen this issue if you have any other questions. thanks!

closed 2022-10-14 by SuZhan0322