Xilinx/Vitis-AI #616: Availability of Pytorch/Python sample code?

larry-rystrom, 2021-12-21:
I apologize in advance if this is not the correct place to ask this question.

I have been given the task of porting our Pytorch UNET convolution model to run on Xilinx, but I see no Pytorch/Python entry in the "VART Samples" table in the readme ( https://github.com/Xilinx/Vitis-AI/tree/master/demo/VART ).

1. Can anyone point me at a Pytorch/Python combination sample, either official or unofficial?
2. Failing that, is there a list of step by step instructions for converting a model from Pytorch to Xilinix? I have the model and model weights as saved by "torch.save(model.state_dict(), PATH)".

Thanks,

Larry

qianglin-xlnx, 2022-01-18:
Hi @larry-rystrom 
You can refer to https://github.com/Xilinx/Vitis-AI-Tutorials/tree/master/Introduction/01-Workflows

larry-rystrom, 2022-01-18:
Hi Quianglin-xlnx,

Thank you for the response. My goal is to evaluate our Deep Learning model on a Xilinx Kria board and obtain basic performance information.  I  have the model and model weights as Pytorch 1.9 python files.

As far as Vitis-AI-Tutorials/Design_Tutorials/09-mnist_pyt/:

  1.  Could you please confirm that the Xilinx Kria is supported? This is not obvious to me from the ReadMe.
  2.  In order to run the model it must first be “quantized” and other steps followed. But I cannot get a Pytorch 1.9 model to load under the 2+ year Pytorch. Instead I get RuntimeError: version_ <= kMaxSupportedFileFormatVersion INTERNAL ASSERT FAILED at ..\caffe2\serialize\inline_container.cc:132, please report a bug to PyTorch. Attempted to read a PyTorch file with version 3, but the maximum supported version for reading is 2. Your PyTorch installation may be too old. (init at ..\caffe2\serialize\inline_container.cc:132) (no backtrace available)
I did have a go at backporting the model from Pytorch 1.10 with the _use_new_zipfile_serialization=False but with no success.

The model was trained for over 100 hours and I cannot easily reproduce it on the older Pytorch.  When will the version of Pytorch be updated?

Thanks,

Larry

CAUTION: This email originated from outside of Thermo Fisher Scientific. If you believe it to be suspicious, report using the Report Phish button in Outlook or send to ***@***.***

Hi @larry-rystrom<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_larry-2Drystrom&d=DwMCaQ&c=q6k2DsTcEGCcCb_WtVSz6hhIl8hvYssy7sH8ZwfbbKU&r=-SqA0CoQEVWMaGfI-1LVoW9VDW0dMfnBP-FfJm6n5D8&m=sjUNCvdG_PwOtp_Pllq1fPxEzdfyrDf07TZ6zrB9_35vomNZWZdiUPfRclmpXuRs&s=dE7x2BTIXz-1RL9qFs_RIVOXAj8XIELuLHhE-hrZV9U&e=>
You can refer to https://github.com/Xilinx/Vitis-AI-Tutorials/tree/master/Introduction/01-Workflows<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_Xilinx_Vitis-2DAI-2DTutorials_tree_master_Introduction_01-2DWorkflows&d=DwMCaQ&c=q6k2DsTcEGCcCb_WtVSz6hhIl8hvYssy7sH8ZwfbbKU&r=-SqA0CoQEVWMaGfI-1LVoW9VDW0dMfnBP-FfJm6n5D8&m=sjUNCvdG_PwOtp_Pllq1fPxEzdfyrDf07TZ6zrB9_35vomNZWZdiUPfRclmpXuRs&s=rSKhKbMypHsXvxTGKnFDIkxXYqTGPWuwZYSBiglbOe0&e=>

qianglin-xlnx, 2022-02-16:
Hi @larry-rystrom 
1 For Vitis-AI, Kria KV260 board is supported.
2 Pytorch 1.9 is supported in VAI2.0. Please have a try. Thank you.

closed 2022-03-01 by qianglin-xlnx