Xilinx/Vitis-AI #617: UnboundLocalError: local variable 'op' referenced before assignment

Ace-cream, 2021-12-23:
I apologize in advance if this is not the correct way to ask this question.
I am preparing to quantify the trained YOLOX model through vai_q_pytorch, but encountered this error:
**Traceback (most recent call last):
  File "yolox_quantify.py", line 236, in <module>
    quantization(model=model, model_path=model_path)
  File "yolox_quantify.py", line 180, in quantization
    quant_mode, model, (input), device=device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 77, in __init__
    custom_quant_ops = custom_quant_ops)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 122, in __init__
    device=device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 175, in prepare_quantizable_module
    graph = parse_module(module, input_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 78, in parse_module
    module, input_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/parser.py", line 71, in __call__
    nndct_graph = self._convert_graph(raw_graph, self._get_device_info(module, input_args))
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/parser.py", line 113, in _convert_graph
    nndct_node.op = op_creator(self, nndct_node)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/op_dispatcher.py", line 106, in __call__
    nndct_node.raw_kind)(*parser.node_input_args[nndct_node])
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/op_dispatcher.py", line 152, in _convolution
    op.set_config('in_channels', weight_size[1] * groups)
UnboundLocalError: local variable 'op' referenced before assignment**

My `quant.py` is like `resnet18_quant.py` and I am sure that my net is ok. I don't know why this problem occurs and how to fix it.
Can anyone help me?
Thanks!

qianglin-xlnx, 2021-12-24:
Hi @Ace-cream 
YOLOX will be supported in the next VAI release. Stay tuned.

Ace-cream, 2021-12-24:
> Hi @Ace-cream YOLOX will be supported in the next VAI release. Stay tuned.

Thanks a lot for your help @qianglin-xlnx.
Would you please tell me when the new VAI will be released if you konw?  It's important for me. Thanks again!

jlamperez, 2022-01-17:
> Hi @Ace-cream YOLOX will be supported in the next VAI release. Stay tuned.

Any info about when the new VAI is going to be released? Is going to support Vitis 2021.2?

qianglin-xlnx, 2022-01-18:
The new VAI will be released soon. Vitis 2021.2 will be supported in new VAI.

qianglin-xlnx, 2022-01-20:
Hi @Ace-cream and @jlamperez 

VAI2.0 is live now. You can have a try. Thank you.

qianglin-xlnx, 2022-03-03:
Hi @Ace-cream 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-03 by qianglin-xlnx

Shreyas-NR, 2022-07-30:
Hi @Ace-cream ,
I am also working on quantifying the YOLOX model through vai_q_pytorch, 
1. Could you share the changes you did to the model architecture to train the model that is acceptable by the DPU?(activation layer, max pool,...)
2. Can you share with me the yolox_quantify.py file that you have written for reference?