Xilinx/Vitis-AI #621: VCK5000 Vivado ES1 Links Permission

ombhilare999, 2021-12-31:
I already got approved for the vck5000 secure site, but the links mentioned in the readme are broken. They redirected to the vck5000 base page.
I even tried with wget for the pdf's but they are not working. Kindly provide these files.

https://github.com/Xilinx/Vitis-AI/blob/122649108b673b6579c86291619b6ec8fd664d35/setup/vck5000/README.md?plain=1#L152-L159

Thanks & Regards,
Omkar Bhilare

fanz-xlnx, 2022-01-27:
Hi Omkar,

If you are using the VCK5000 ES card. Please refer to the Vitis AI 1.4.1 release:
https://github.com/Xilinx/Vitis-AI/tree/1.4/setup/vck5000

for VCK5000 prod card, please refer to the 2.0 release.
https://github.com/Xilinx/Vitis-AI/tree/master/setup/vck5000

May I know which pdf you are looking for?

SuZhan0322, 2022-09-01:
Hi @ombhilare999, It looks like you have found the tutorial for vck5000, please reopen this issue if you have any other questions. thanks！

closed 2022-09-01 by SuZhan0322