Xilinx/Vitis-AI #623: Check failed: !the_factory_methods.empty() KV260

jackbonnellssa, 2022-01-04:
Hi i am struggling to run a custom DPU application through xmutil on petalinux 2021.2. I have attached my application. when loading the application i get this output:

```
xilinx-k26-som-2021_2:~/Vitis-AI/demo/VART/resnet50$ sudo xmutil loadapp user-firmware
DFX-MGRD> daemon loading accel user-firmware 
 
Jan  3 04:56:56 xilinx-k26-som-2021_2 kernel: [ 1362.791145] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /fpga-full/fpga-config-from-dmabuf
Jan  3 04:56:56 xilinx-k26-som-2021_2 kernel: [ 1362.791159] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /fpga-full/firmware-name
Jan  3 04:56:56 xilinx-k26-som-2021_2 kernel: [ 1362.791165] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /fpga-full/resets
Jan  3 04:56:56 xilinx-k26-som-2021_2 kernel: [ 1362.791305] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/overlay0
Jan  3 04:56:56 xilinx-k26-som-2021_2 kernel: [ 1362.791320] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/overlay1
Jan  3 04:56:56 xilinx-k26-som-2021_2 kernel: [ 1362.791335] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/afi0
Jan  3 04:56:56 xilinx-k26-som-2021_2 kernel: [ 1362.791351] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/clocking0
Jan  3 04:56:56 xilinx-k26-som-2021_2 kernel: [ 1362.791366] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/overlay2
Jan  3 04:56:56 xilinx-k26-som-2021_2 kernel: [ 1362.791381] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/DPUCZDX8G_0
Jan  3 04:56:56 xilinx-k26-som-2021_2 kernel: [ 1362.791396] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/misc_clk_0
DFX-MGRD> Successfully loaded base design.
 
Accelerator loaded to slot 0
Socket 6 closed by client
```
my lsmod output is:

```
lsmod
Module                  Size  Used by
usb5744                16384  0
dmaproxy               16384  0
mali                  233472  0
uio_pdrv_genirq        16384  0

if i run example vitis ai resnet50 i get the output:
```

```
sudo ./resnet50 /usr/share/vitis_ai_library/models/resnet50.xmodel
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0103 04:57:04.826944  1532 main.cc:292] create running for subgraph: subgraph_conv1
F0103 04:57:04.844861  1532 dpu_controller.cpp:44] Check failed: !the_factory_methods.empty() 
*** Check failure stack trace: ***
Aborted
```
**packagegroup-petalinux-vitisai** is enabled in petalinux.

 

I have also loaded the build **target_vart_setup.sh**

 

My block Design is attached and so is my application -> [user-firmware.zip](https://github.com/Xilinx/Vitis-AI/files/7807056/user-firmware.zip)

[image]

andrew0917, 2022-01-12:
Hi @jackbonnellssa 
       I don't  konw where and how do you build your xilinx-k26-som rootfs system in 2021.2 version, and I found that there is no zocl driver in your `lsmod` command output, in my opinion, you need to add zocl driver to your rootfs system. by the way, zocl is also used for XRT, so your rootfs system doesn't enable xrt either. you can run `petalinux-config -c rootfs` command to enable zocl and xrt into your rootfs system.

qianglin-xlnx, 2022-03-14:
Hi @jackbonnellssa 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-14 by qianglin-xlnx

iivanov1996, 2022-04-11:
Hi @qianglin-xlnx,

I can see that I have installed ZOCL and XRT on my system.
[image]

However, I am getting the same error message when running the model. The lsmod output follows:
[image]

If the drivers are not installed properly, how to install them?

masommao, 2022-05-23:
I Running the command“./demo_yolov3 /usr/share/vitis_ai_library/models/yolov3_voc/yolov3_voc.xmodel \
    demo_yolov3.jpg”error“Check failed: !the_factory_methods.empty() ”
but running “sh build.sh”is ok
[image]
。

iraadit, 2023-09-13:
Hi @iivanov1996,

Have you ever resolved this issue?

I ran in exactly the same error with Vitis-AI 3.5 and KV260.

Thank you