Xilinx/Vitis-AI #624: Failed to run tf_resnet example on AWS f1 (DPUCADF8H)

HasinduKariyawasam, 2022-01-05:
After setting up Vitis AI on AWS f1.2xlarge instance, I tried to run the tf_resnet example. I managed to follow all the steps until the final one. First, I got the following error.
```console
(vitis-ai-tensorflow) Vitis-AI /workspace/examples/DPUCADF8H/tf_resnet > ./resnet_example ./out/resnet50_v1_tf_compiled.xmodel ./images/
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0104 23:07:24.276937   464 main.cc:293] create running for subgraph: subgraph_resnet_v1_50/block1/unit_1/bottleneck_v1/add
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::property_tree::json_parser::json_parser_error> >'
  what():  <unspecified file>(1): expected value
Aborted (core dumped)
```

Then, I tried running the command with `sudo` and I got a new error.
```console
(vitis-ai-tensorflow) Vitis-AI /workspace/examples/DPUCADF8H/tf_resnet > sudo ./resnet_example ./out/resnet50_v1_tf_compiled.xmodel ./images/
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0104 23:11:30.517123   466 main.cc:293] create running for subgraph: subgraph_resnet_v1_50/block1/unit_1/bottleneck_v1/add
terminate called after throwing an instance of 'std::runtime_error'
  what():  Error: xclbinPath is not set, please consider setting XLNX_VART_FIRMWARE.
Aborted
```
When I checked the `XLNX_VART_FIRMWARE` variable, it is already set as shown below.
```console
(vitis-ai-tensorflow) Vitis-AI /workspace/examples/DPUCADF8H/tf_resnet > echo $XLNX_VART_FIRMWARE 
/opt/xilinx/overlaybins/DPUCADF8H/dpu-aws.xclbin
``` 
I am not sure why this error occurs. Any pointers to fix this issue are highly appreciated.

bryanloz-xilinx, 2022-01-18:
If you see an error about  XLNX_VART_FIRMWARE not being set, it isn’t set. The code will not lie. When you became root, you likely lost that env variable.

Focus on the first error, something threw an exception. Try to figure out what piece of code threw that. Also, verify that the xclbin is downloaded, and is readable at the location specified by the aforementioned env variable.

China-CC-lon, 2022-02-22:
@HasinduKariyawasam Did you solve this problem please? I also had the same problem

HasinduKariyawasam, 2022-02-22:
@China-CC-lon unfortunately not. I still couldn't get through this issue.
@bryanloz-xilinx thank you for your answer. The xclbin file is there in the correct location. I still couldn't figure out which part of the code throwing the error. I was wondering whether I need to use the awsxclbin file instead of xclbin, since I am using AWS F1 instance.

China-CC-lon, 2022-02-22:
@bryanloz-xilinx I am using VCK5000, boost error is not a user permissions problem? I added sudo to the permission and then said that the environment variable problem, whether to enter the root user, then enter the docker container, and then run demo.

China-CC-lon, 2022-02-23:
@HasinduKariyawasam I solved this problem. I ran the install.sh script on the official website again, reduced the XRM version to version 1.2.1539 instead of version 1.3.29, and then redeployed the development platform environment, and then just followed the process.

HasinduKariyawasam, 2022-02-23:
That's great. I'll try the same. Thank you @China-CC-lon

RenzoPederzoli, 2022-03-28:
[image]
Hello, I am also running on the same environment as op. But instead I get no error, only that it seems stuck.
Has anyone else experienced this? Any advice would be greatly appreaciated.

Thanks

HasinduKariyawasam, 2022-03-30:
Hi @RenzoPederzoli,
I solved this issue. What I did is I set up Vitis AI in the F1 instance again as described [here](https://github.com/parimalp/xup-vitis-ai-tutorial/blob/main/docs/setup_vitisai_awsf1.md). Hope this helps!

closed 2022-09-01 by SuZhan0322