Xilinx/Vitis-AI #625: Errors got when compiling vart xir "undefined reference to ..."

xxliu1996, 2022-01-06:
I am trying to compile the vitis ai quantizer tool from source code. And I found that I need to compile and install unilog and xir first. So, I go to the directory of xir and follow the instructions. But I got error which seems related to the conflicts of "c++11" and "c++17. Here are some of the error messages:
-------------------------------------------------------------
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1820: undefined reference to `ErrorCode::getErrDebugInfo[abi:cxx11]() const'
CMakeFiles/xir.dir/op/shape_inference.cpp.o: In function `xir::shape_infer_ddr_flatten_concat(xir::Op*)':
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1854: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1854: undefined reference to `UniLog::UniLogPrefix[abi:cxx11]'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1854: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1854: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1854: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1854: undefined reference to `ErrorCode::getErrDebugInfo[abi:cxx11]() const'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1860: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1860: undefined reference to `UniLog::UniLogPrefix[abi:cxx11]'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1860: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1860: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1860: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1860: undefined reference to `ErrorCode::getErrDebugInfo[abi:cxx11]() const'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1871: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1871: undefined reference to `UniLog::UniLogPrefix[abi:cxx11]'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1871: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1871: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1871: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1871: undefined reference to `ErrorCode::getErrDebugInfo[abi:cxx11]() const'
CMakeFiles/xir.dir/op/shape_inference.cpp.o: In function `xir::shape_infer_reshape_fix(xir::Op*)':
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to `UniLog::UniLogPrefix[abi:cxx11]'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to `ErrorCode::getErrDebugInfo[abi:cxx11]() const'
CMakeFiles/xir.dir/op/shape_inference.cpp.o: In function `xir::flatten_dims(std::vector<int, std::allocator<int> > const&, int const&, int const&)':
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1912: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1912: undefined reference to `UniLog::UniLogPrefix[abi:cxx11]'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1912: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1912: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1912: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1912: undefined reference to `ErrorCode::getErrDebugInfo[abi:cxx11]() const'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1915: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1915: undefined reference to `UniLog::UniLogPrefix[abi:cxx11]'
[... cut ...]
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/tensor/tensor_imp.cpp:181: undefined reference to `ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/tensor/tensor_imp.cpp:181: undefined reference to `ErrorCode::getErrDebugInfo[abi:cxx11]() const'
collect2: error: ld returned 1 exit status
make[2]: *** [src/xir/libxir.so.1.3.2] Error 1
make[1]: *** [src/xir/CMakeFiles/xir.dir/all] Error 2
make: *** [all] Error 2
-------------------------------------------------------------------

xxliu1996, 2022-01-06:
I have tried different gcc versions, also tried to change the CMakeList.txt file, like change cxx_17 to cxx_11, but new error occurs. Also, I have tried to add `--cmake-options -DCMAKE_CXX_FLAGS='-D_GLIBCXX_USE_CXX11_ABI=0'` at the end of my cmake.sh command, but no help to solve this problem. Anyone have got the same issues?

xlinx-dachang, 2022-01-07:
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

Give it a chance to add set(CMAKE_CXX_STANDARD 17)<http://www.cmake.org/cmake/help/v3.1/variable/CMAKE_CXX_STANDARD.html> to the CMakeLists.txt.

/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1854: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' /Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1854: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1860: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' /Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1860: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1871: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' /Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1871: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)'
CMakeFiles/xir.dir/op/shape_inference.cpp.o: In function xir::shape_infer_reshape_fix(xir::Op*)': /Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to UniLog::UniLogPrefix[abi:cxx11]' /Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' /Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1898: undefined reference to ErrorCode::getErrDebugInfo[abi:cxx11]() const' CMakeFiles/xir.dir/op/shape_inference.cpp.o: In function xir::flatten_dims(std::vector<int, std::allocator > const&, int const&, int const&)':
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1912: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' /Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1912: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1915: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' /Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:1915: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)'
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:153: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' /Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/op/shape_inference.cpp:153: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)'
CMakeFiles/xir.dir/graph/graph_imp.cpp.o: In function xir::GraphImp::add_op(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unique_ptr<xir::Attrs, std::default_delete<xir::Attrs> >, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<xir::Op*, std::allocator<xir::Op*> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::vector<xir::Op*, std::allocator<xir::Op*> > > > > const&, xir::DataType co [... cut ...]
/Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/tensor/tensor_imp.cpp:181: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' /Users/xliu254/Research/Software/src/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/tensor/tensor_imp.cpp:181: undefined reference to ErrorCodeFactory::genErrorCode(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)'

xxliu1996, 2022-01-09:
@xlinx-dachang Thanks. I tried, but it seems not working. Acutally, I am setting up the environment on a HPC, so I do not have full authority of installing specific dependent packages. Here is a list of package versios I use.
python/3.7.4_gcc-8.3.0
cmake/3.16.2_gcc-8.3.0
gcc/9.2.0
protobuf/3.6.1

kmhatre14, 2023-12-04:
Is there any update on this issue? I am trying to setup XIR on my side and facing the same issue