Xilinx/Vitis-AI #626: Tensorflow Quantiser - Issue with Input shape

kavyab11, 2022-01-06:
Hi, I am trying to use the vai tensorflow quantiser to quantise a custom model consisting of fully connected layers. The model has multiple input nodes whose shapes are 2 dimensional (like ?,400 : ?,100 : ?,100).

I have tried using both tensorflow and tensorflow2 quantisers.
- vai_q_tensorflow issue: Accepts only 4 dimensional inputs
- vai_q_tensorflow2 issue: Accepts 2 dimensional input but only accepts a single input node

I have ensured that I am using the updated versions of all tools in Vitis AI. I am using the vai1.2 version from Vitis AI 1.4. I would like to know if there is any way to quantise a model which has multiple 2-dimensional input nodes which are passed on to dense layers consisting of 2D weight matrices, like in the example mentioned above.

Issue #57 refers to the input shape issue, but I am unable to pass on 2D inputs in the latest version as well.
Thanks!

xieyi4650, 2022-01-18:
Hi @kavyab11 ,
the dpu input node must be 4 dimensional,  so maybe you can add some expand, squeeze or reshape op to operate the input tensor or tensor before enter "input_node".

qianglin-xlnx, 2023-04-27:
Closing since no activity for more than 1 year, please open a new issue if you still have any questions, thanks.

closed 2023-04-27 by qianglin-xlnx