Xilinx/Vitis-AI #627: Using VCK5000 Development platform file for Vitis application project

elliottbinder, 2022-01-10:
Is there any reason I shouldn't be able to use the development platform package from the Vitis AI VCK5000 setup page to program the card with Vitis? I've tried building the vector addition example in Vitis for the VCK5000 hardware target, but I'm getting errors. I'm not sure if this is related to the platform files or something else.

More detail can be found here: https://support.xilinx.com/s/question/0D52E00006vGY0BSAW/vitis-vector-addition-example-for-vck5000-module-axiregisterslicev2122axiregisterslice-not-found?language=en_US

hanxue, 2022-03-17:
Hi @elliottbinder , this seems to be a question on building Vector Addition on Vitis for VCK5000. I see that Grace and Emery has been following up at https://support.xilinx.com/s/question/0D52E00006vGY0BSAW/vitis-vector-addition-example-for-vck5000-module-axiregisterslicev2122axiregisterslice-not-found?language=en_US 

Is there anything Vitis AI related that needs to be address?

SuZhan0322, 2022-09-01:
Hi @elliottbinder , Please refer the VCK5000 tutorial for VAI2.5 release. And please reopen the issue if you have any other questions. thanks!

closed 2022-09-01 by SuZhan0322