Xilinx/Vitis-AI #630: Support of tensorflow >2.3

Laadr, 2022-01-12:
My team and I are encountering a few issue with TF2.3 and would like to upgrade to a more recent version but still be compatible with VITIS AI.
However, before taking this decision, we are wondering why VITIS AI chose to support specifically TF2.3. Is there a specific issue with more recent versions? or is it only because these versions have not yet been checked?
Thank in advance for your answer.

jeinstei, 2022-01-12:
Have you tried rebuilding the Dockerfile with a later TF? My models aren't working at all already, but I would be curious if you update it and build and if it still works for you. I'm definitely curious...

Laadr, 2022-01-14:
After rethinking my question, I think it was a bit off. I was referring to the TF version we use to generate the models that we then process with the tools of vitis ai.  However, the version of TF specified in the documentation is referring to the version of the modified version of TF made for VITIS AI. This means that as long as my protobuf model can be read by TF2.3, everything should be fine.

@jeinstei : If you are talking about VITIS AI docker, I think rebuilding the Dockerfile with a later TF is not possible. The docker builds the xilinx version of TF and not a conventional TF and there is no newer version than the one based on TF2.3 codes I think.

I hope I haven't wasted anyone's time and I will try to think more before asking a question next time!

closed 2022-01-14 by Laadr