Xilinx/Vitis-AI #632: vaitrace power consumption (U50 board)

vittofa, 2022-01-13:
Hello,
Is it possible with "vaitrace" to trace the board power consumption running my neural network? (I'm using U50 board)
If not, what's the best approach to do it?

Thanks.

li900309, 2022-01-18:
hi @vittofa , the **xbutil query** command could be help, just setup XRT environment(source PATH_TO_XRT/setup.sh), then run **xbutil query** you can get the power and temp info from the report 
[image]

vittofa, 2022-01-25:
Thank you @li900309, I already use this command, I was wondering if could be possible to have a more accurate way to measure power consumption, but I guess I'll have to deal with it.

vittofa, 2022-02-21:
For those interested in a complete prifiler output with power consmption for python's scripts you need to use "vaitrace" with "xrt.ini", you need to follow this instruction:
https://xilinx.github.io/XRT/master/html/xrt_ini.html
and set the desired parameters... 
https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/xrt.ini-File
Thanks, bye.

closed 2022-02-21 by vittofa